gazebosim/gazebo_test_cases #467: Include arm64 machine as a supported platform

ARLunan, 2023-09-03:
Required Info:

Operating System:
Ubuntu 22.04.3 arm64
ROS2 Version:
ROS Iron binary
Version or commit hash:
1.2.2-1jammy.20230804.210746

## Desired behavior
Install Gazebo Harmonic  on arm64 paltforms such as MAC M1 used as a ROS 2 Desktop

Voldivh, 2023-09-04:
Hi @ARLunan will you be working on this issue?

ARLunan, 2023-09-04:
Hi & tnx for reply. No I cannot do that as I am not familiar with this type of code development and release process into repositories. I would be able to load and test released code on an arm64 machine.

azeey, 2023-09-13:
@ARLunan There's some work underway in https://github.com/osrf/homebrew-simulation/pull/2273 regarding support for M1 machines. I recommend following up there. I'll go ahead and close this issue.

closed 2023-09-13 by azeey

ARLunan, 2023-09-13:
> @ARLunan <https://github.com/ARLunan> There's some work underway in osrf/homebrew-simulation#2273 <https://github.com/osrf/homebrew-simulation/pull/2273> regarding support for M1 machines. I recommend following up there. I'll go ahead and close this issue.

Addisu, My Issue needs a clarification, in that the application is NOT native MAC functionality. My request is for support of a standard Ubuntu 22.04 OS platform that is installed on a Virtual Machine on the MAC M1, which is a very effective ROS Desktop.  The package binaries need to be arm64 due the underlying MAC M1 processor. 

This is a similar situation to most of the existing ROS 2 Packages are available that are intended to by installed on either Ubuntu 22.04 amd64 for Intel PC hardware or arm64 for MAC M1 or Video Machines.  

Again, no issue to not providing installations on Native MAC so the “homebrew” work is not relevant to my issue.