gazebosim/gazebo_test_cases #468: Unable to Install nav2-ros-iron-bringup using Gazebo Harmonic due missing gazebo-ros-pkgs and turtlebot3-gazebo

ARLunan, 2023-09-03:
Required Info:
Operating System: Ubuntu 22.04.3 arm64 and amd64
ROS2 Version: ROS Iron binary
Version or commit hash: 1.2.2-1jammy.20230804.210746

Steps to reproduce issue
Try to install ros-iron-nav2-bringup on arm64. install works on amd64 due Gazebo Classic is available on this platform.

Expected behavior
ros-iron-nav2-bringup using Gazebo Harmonic should be available for arm64 and amd64

Actual behavior
ros-iron-nav2-bringup 1.2.2-1 deb package for arm64 doesn't exist in http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-iron-nav2-bringup/

## Implementation suggestion
Provide ROS 2 Iron Gazebo Harmonic functional equivalent gazebo-ros-pkgs and turtlebot3-gazebo to support ROS Iron navigation2/nav2-bringup. For ref: [turtlebot3_simulations ](https://github.com/ROBOTIS-GIT/turtlebot3_simulations  )

ARLunan, 2023-09-05:
Revised "Steps to reproduce issue" to report the Issue also occurs on amd64 Intel PC machine.
Try to install **ros-iron-nav2-bringup** and **ros-iron-nav2-system-tools** on arm64 (MAC M1 VM)and amd64 (IntelPC) Ubuntu 22.04.03 machines. Due to inclusion of <build_depend>gazebo_ros_pkgs</build_depend> and <exec_depend>turtlebot3_gazebo</exec_depend> which are Gazebo Classic Packages. 

Terminal log in both cases:
ubuntu@7400-ub22i-r2:~$ sudo apt install ros-iron-nav2-system-tests
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help resolve the situation:

The following packages have unmet dependencies:
 gz-tools2 : Conflicts: gazebo (>= 11.0.0) but 11.10.2+dfsg-1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
ubuntu@7400-ub22i-r2:~$ 

**Actual Behavior**
ros-iron-nav2-bringup 1.2.2-1 deb package for arm64 doesn't exist in [nav2-bringup](http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-iron-nav2-bringup/) doesn't exist AND
ros-iron-nav2-system-tests.2.2-1 deb package for amd64 will not install  [nav2_system_tests](http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-iron-nav2-system-tests/)

**Implementation** 
Provide ROS 2 Iron Gazebo Harmonic functional equivalent gazebo-ros-pkgs and turtlebot3-gazebo to support amd64 and arm64 ROS Iron **navigation2/nav2-bringup** and **navigation2/nav2-system-tests**. For ref: [turtlebot3_simulations](https://github.com/ROBOTIS-GIT/turtlebot3_simulations)

azeey, 2023-09-13:
The `ros_gz` binaries for the Iron/Harmonic pairing are still under testing and are not ready yet (see https://github.com/gazebo-tooling/release-tools/issues/994). I believe nav2 currently only supports Gazebo classic. There's currently a [PR](https://github.com/ros-planning/navigation2/pull/3634) open for adding support for the new Gazebo, so that would be the best place to check. /cc @ahcorde 

Since this is more a feature request for nav2, I'lll go ahead and close this issue.

closed 2023-09-13 by azeey