gazebosim/gazebo_test_cases #528: gz-common: Installation

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/common/5/install.html)
- [Source](https://github.com/gazebosim/gz-common/tree/gz-common5/tutorials/install.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-13:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Details

Continuing from
- https://github.com/gazebosim/gazebo_test_cases/issues/945#issuecomment-1718203548
- https://github.com/gazebosim/gazebo_test_cases/issues/600#issuecomment-1718340446
 
`gz-common5` builds with 1 warning

```bash
% colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-select gz-common5
Starting >>> gz-common5
--- stderr: gz-common5                                
CMake Warning (dev) at /Users/rhys/Code/osrf/gz_harmonic_ws/install/share/cmake/gz-cmake3/cmake3/GzFindPackage.cmake:456 (message):
   -- THIS MESSAGE IS INTENDED FOR GZ-COMMON AUTHORS --
      (IF YOU SEE THIS, PLEASE REPORT IT)

  Could not find pkg-config information for UUID.  It was not provided by the
  find-module for the package, nor was it explicitly passed into the call to
  gz_find_package(~).  This is most likely an error in this project's use of
  gz-cmake.
Call Stack (most recent call first):
  CMakeLists.txt:81 (gz_find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< gz-common5 [10.5s]

Summary: 1 package finished [10.7s]
  1 package had stderr output: gz-common5
```

Also builds using

```bash
cd ./src/gz-common && mkdir build && cd build
cmake .. && make -j8
```

Tests

```bash
% make test
Running tests...
Test project /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-common/build
        Start   1: UNIT_Base64_TEST
  1/125 Test   #1: UNIT_Base64_TEST ............................   Passed    0.51 sec
        Start   2: check_UNIT_Base64_TEST
  2/125 Test   #2: check_UNIT_Base64_TEST ......................   Passed    0.03 sec
        Start   3: UNIT_Battery_TEST
  3/125 Test   #3: UNIT_Battery_TEST ...........................   Passed    0.13 sec
        Start   4: check_UNIT_Battery_TEST
  4/125 Test   #4: check_UNIT_Battery_TEST .....................   Passed    0.03 sec
        Start   5: UNIT_Console_TEST
  5/125 Test   #5: UNIT_Console_TEST ...........................   Passed    0.14 sec
        Start   6: check_UNIT_Console_TEST
  6/125 Test   #6: check_UNIT_Console_TEST .....................   Passed    0.02 sec
        Start   7: UNIT_EnumIface_TEST
  7/125 Test   #7: UNIT_EnumIface_TEST .........................   Passed    0.12 sec
        Start   8: check_UNIT_EnumIface_TEST
[... 215 lines cut ...]
        Start 116: UNIT_Profiler_Remotery_TEST
116/125 Test #116: UNIT_Profiler_Remotery_TEST .................   Passed    0.13 sec
        Start 117: check_UNIT_Profiler_Remotery_TEST
117/125 Test #117: check_UNIT_Profiler_Remotery_TEST ...........   Passed    0.03 sec
        Start 118: UNIT_AutoLogFixture_TEST
118/125 Test #118: UNIT_AutoLogFixture_TEST ....................   Passed    0.12 sec
        Start 119: check_UNIT_AutoLogFixture_TEST
119/125 Test #119: check_UNIT_AutoLogFixture_TEST ..............   Passed    0.03 sec
        Start 120: UNIT_CMakeTestPaths_TEST
120/125 Test #120: UNIT_CMakeTestPaths_TEST ....................   Passed    0.12 sec
        Start 121: check_UNIT_CMakeTestPaths_TEST
121/125 Test #121: check_UNIT_CMakeTestPaths_TEST ..............   Passed    0.03 sec
        Start 122: UNIT_RedirectConsoleStream_TEST
122/125 Test #122: UNIT_RedirectConsoleStream_TEST .............   Passed    0.12 sec
        Start 123: check_UNIT_RedirectConsoleStream_TEST
123/125 Test #123: check_UNIT_RedirectConsoleStream_TEST .......   Passed    0.03 sec
        Start 124: UNIT_Utils_TEST
124/125 Test #124: UNIT_Utils_TEST .............................   Passed    0.12 sec
        Start 125: check_UNIT_Utils_TEST
125/125 Test #125: check_UNIT_Utils_TEST .......................   Passed    0.03 sec

100% tests passed, 0 tests failed out of 125

Total Test time (real) =  27.90 sec
```

Voldivh, 2023-09-14:
Thanks @srmainwaring

closed 2023-09-14 by Voldivh