gazebosim/gazebo_test_cases #546: gz-fuel-tools: Installation

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/fuel_tools/9/install.html)
- [Source](https://github.com/gazebosim/gz-fuel-tools/tree/gz-fuel-tools9/tutorials/01_installation.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-13:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Suggested changes

The instructions refer to the wrong branch

```diff
- git clone https://github.com/gazebosim/gz-fuel-tools -b ign-fuel-tools<#>
+ git clone https://github.com/gazebosim/gz-fuel-tools -b gz-fuel-tools<#>
```

### Details

Continuing from

- https://github.com/gazebosim/gazebo_test_cases/issues/945#issuecomment-1718203548
- https://github.com/gazebosim/gazebo_test_cases/issues/900#issuecomment-1718362032
- https://github.com/gazebosim/gazebo_test_cases/issues/600#issuecomment-1718340446
- https://github.com/gazebosim/gazebo_test_cases/issues/528#issuecomment-1718347066
- https://github.com/gazebosim/gazebo_test_cases/issues/621#issuecomment-1718380277

`gz-fuel_tools9` builds clean

```bash
% colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-select gz-fuel_tools9
Starting >>> gz-fuel_tools9
Finished <<< gz-fuel_tools9 [6.52s]                     

Summary: 1 package finished [6.92s]
```

Also builds using

```bash
cd ./src/gz-fuel-tools && mkdir build && cd build
cmake .. && make -j8
```

Test

```bash
% make test
Running tests...
Test project /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-fuel-tools/build
      Start  1: UNIT_ClientConfig_TEST
 1/36 Test  #1: UNIT_ClientConfig_TEST .................   Passed    0.66 sec
      Start  2: check_UNIT_ClientConfig_TEST
 2/36 Test  #2: check_UNIT_ClientConfig_TEST ...........   Passed    0.02 sec
      Start  3: UNIT_CollectionIdentifier_TEST
 3/36 Test  #3: UNIT_CollectionIdentifier_TEST .........   Passed    0.15 sec
      Start  4: check_UNIT_CollectionIdentifier_TEST
 4/36 Test  #4: check_UNIT_CollectionIdentifier_TEST ...   Passed    0.02 sec
      Start  5: UNIT_FuelClient_TEST
 5/36 Test  #5: UNIT_FuelClient_TEST ...................   Passed   26.07 sec
      Start  6: check_UNIT_FuelClient_TEST
 6/36 Test  #6: check_UNIT_FuelClient_TEST .............   Passed    0.04 sec
      Start  7: UNIT_gz_src_TEST
 7/36 Test  #7: UNIT_gz_src_TEST .......................   Passed   27.72 sec
      Start  8: check_UNIT_gz_src_TEST
 8/36 Test  #8: check_UNIT_gz_src_TEST .................   Passed    0.05 sec
      Start  9: UNIT_Interface_TEST
 9/36 Test  #9: UNIT_Interface_TEST ....................   Passed    4.36 sec
      Start 10: check_UNIT_Interface_TEST
10/36 Test #10: check_UNIT_Interface_TEST ..............   Passed    0.04 sec
      Start 11: UNIT_Helpers_TEST
11/36 Test #11: UNIT_Helpers_TEST ......................   Passed    0.20 sec
      Start 12: check_UNIT_Helpers_TEST
12/36 Test #12: check_UNIT_Helpers_TEST ................   Passed    0.02 sec
      Start 13: UNIT_JSONParser_TEST
13/36 Test #13: UNIT_JSONParser_TEST ...................   Passed    0.18 sec
      Start 14: check_UNIT_JSONParser_TEST
14/36 Test #14: check_UNIT_JSONParser_TEST .............   Passed    0.02 sec
      Start 15: UNIT_LocalCache_TEST
15/36 Test #15: UNIT_LocalCache_TEST ...................   Passed    0.21 sec
      Start 16: check_UNIT_LocalCache_TEST
16/36 Test #16: check_UNIT_LocalCache_TEST .............   Passed    0.02 sec
      Start 17: UNIT_ModelIdentifier_TEST
17/36 Test #17: UNIT_ModelIdentifier_TEST ..............   Passed    0.18 sec
      Start 18: check_UNIT_ModelIdentifier_TEST
18/36 Test #18: check_UNIT_ModelIdentifier_TEST ........   Passed    0.02 sec
      Start 19: UNIT_ModelIter_TEST
19/36 Test #19: UNIT_ModelIter_TEST ....................   Passed    0.17 sec
      Start 20: check_UNIT_ModelIter_TEST
20/36 Test #20: check_UNIT_ModelIter_TEST ..............   Passed    0.02 sec
      Start 21: UNIT_Model_TEST
21/36 Test #21: UNIT_Model_TEST ........................   Passed    0.16 sec
      Start 22: check_UNIT_Model_TEST
22/36 Test #22: check_UNIT_Model_TEST ..................   Passed    0.02 sec
      Start 23: UNIT_RestClient_TEST
23/36 Test #23: UNIT_RestClient_TEST ...................   Passed    0.15 sec
      Start 24: check_UNIT_RestClient_TEST
24/36 Test #24: check_UNIT_RestClient_TEST .............   Passed    0.02 sec
      Start 25: UNIT_Result_TEST
25/36 Test #25: UNIT_Result_TEST .......................   Passed    0.16 sec
      Start 26: check_UNIT_Result_TEST
26/36 Test #26: check_UNIT_Result_TEST .................   Passed    0.02 sec
      Start 27: UNIT_WorldIdentifier_TEST
27/36 Test #27: UNIT_WorldIdentifier_TEST ..............   Passed    0.17 sec
      Start 28: check_UNIT_WorldIdentifier_TEST
28/36 Test #28: check_UNIT_WorldIdentifier_TEST ........   Passed    0.02 sec
      Start 29: UNIT_WorldIter_TEST
29/36 Test #29: UNIT_WorldIter_TEST ....................   Passed    0.17 sec
      Start 30: check_UNIT_WorldIter_TEST
30/36 Test #30: check_UNIT_WorldIter_TEST ..............   Passed    0.02 sec
      Start 31: UNIT_Zip_TEST
31/36 Test #31: UNIT_Zip_TEST ..........................   Passed    0.16 sec
      Start 32: check_UNIT_Zip_TEST
32/36 Test #32: check_UNIT_Zip_TEST ....................   Passed    0.02 sec
      Start 33: UNIT_gz_TEST
33/36 Test #33: UNIT_gz_TEST ...........................   Passed    0.16 sec
      Start 34: check_UNIT_gz_TEST
34/36 Test #34: check_UNIT_gz_TEST .....................   Passed    0.02 sec
      Start 35: INTEGRATION_zip
35/36 Test #35: INTEGRATION_zip ........................   Passed    0.16 sec
      Start 36: check_INTEGRATION_zip
36/36 Test #36: check_INTEGRATION_zip ..................   Passed    0.02 sec

100% tests passed, 0 tests failed out of 36

Total Test time (real) =  61.63 sec
```

Voldivh, 2023-09-15:
I'll be closing this as the PR with the suggested changes have been merged. Thanks @srmainwaring

closed 2023-09-15 by Voldivh