gazebosim/gazebo_test_cases #585: gz-launch: Installation

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/launch/7/install.html)
- [Source](https://github.com/gazebosim/gz-launch/tree/gz-launch7/tutorials/install.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Suggested changes

Update branch names: `ign` => `gz`

```diff
- These instructions are for installing only Gazebo Launch. If you're interested in using all the Gazebo libraries, not only Igniton Launch, check out this [Gazebo installation](https://gazebosim.org/docs/latest/install).
+ These instructions are for installing only Gazebo Launch. If you're interested in using all the Gazebo libraries, not only Gazebo Launch, check out this [Gazebo installation](https://gazebosim.org/docs/latest/install).
```

```diff
- git clone https://github.com/gazebosim/gz-launch -b ign-launch<#>
+ git clone https://github.com/gazebosim/gz-launch -b gz-launch<#>
```

### Details

Continuing from

- https://github.com/gazebosim/gazebo_test_cases/issues/804#issuecomment-1718954631

`gz-launch` builds

```bash
% colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-select gz-launch7
Starting >>> gz-launch7
--- stderr: gz-launch7                                
CMake Warning at /Users/rhys/Code/osrf/gz_harmonic_ws/install/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:68 (message):
   CONFIGURATION WARNINGS:
   -- Unable to find libwebsockets. The websocket_server plugin will not be built.
Call Stack (most recent call first):
  CMakeLists.txt:111 (gz_configure_build)

---
Finished <<< gz-launch7 [10.5s]

Summary: 1 package finished [11.1s]
  1 package had stderr output: gz-launch7
```

Also build using

```bash
cd ./src/gz-launch && mkdir build && cd build
cmake .. && make -j8
```

Test - clear

```bash
% make test
Running tests...
Test project /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-launch/build
    Start 1: UNIT_Manager_TEST
1/6 Test #1: UNIT_Manager_TEST ..................   Passed    0.51 sec
    Start 2: check_UNIT_Manager_TEST
2/6 Test #2: check_UNIT_Manager_TEST ............   Passed    0.03 sec
    Start 3: UNIT_gz_TEST
3/6 Test #3: UNIT_gz_TEST .......................   Passed    0.88 sec
    Start 4: check_UNIT_gz_TEST
4/6 Test #4: check_UNIT_gz_TEST .................   Passed    0.02 sec
    Start 5: INTEGRATION_faulty_plugins
5/6 Test #5: INTEGRATION_faulty_plugins .........   Passed    0.26 sec
    Start 6: check_INTEGRATION_faulty_plugins
6/6 Test #6: check_INTEGRATION_faulty_plugins ...   Passed    0.03 sec

100% tests passed, 0 tests failed out of 6

Total Test time (real) =   1.72 sec
```

Voldivh, 2023-09-14:
Thanks @srmainwaring I'll be closing this issue since the PR with the suggested changes have been merged.

closed 2023-09-14 by Voldivh