gazebosim/gazebo_test_cases #600: gz-math: Installation

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/math/7/install.html)
- [Source](https://github.com/gazebosim/gz-math/tree/gz-math7/tutorials/installation.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-13:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Details

- Continuing from https://github.com/gazebosim/gazebo_test_cases/issues/945#issuecomment-1718203548

`gz-math7` builds with some warnings

```bash
% colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-select gz-math7
Starting >>> gz-math7
[Processing: gz-math7]                             
--- stderr: gz-math7                                
CMake Warning (dev) at CMakeLists.txt:103 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

warning: unknown warning option '-Wno-maybe-uninitialized'; did you mean '-Wno-uninitialized'? [-Wunknown-warning-option]
warning: unknown warning option '-Wno-maybe-uninitialized'; did you mean '-Wno-uninitialized'? [-Wunknown-warning-option]
warning: unknown warning option '-Wno-maybe-uninitialized'; did you mean '-Wno-uninitialized'? [-Wunknown-warning-option]
warning: unknown warning option '-Wno-maybe-uninitialized'; did you mean '-Wno-uninitialized'? [-Wunknown-warning-option]
warning: unknown warning option '-Wno-maybe-uninitialized'; did you mean '-Wno-uninitialized'? [-Wunknown-warning-option]
warning: unknown warning option '-Wno-maybe-uninitialized'; did you mean '-Wno-uninitialized'? [-Wunknown-warning-option]
/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-math7/lib/ruby/GaussMarkovProcessRUBY_wrap.cxx:1157:3: warning: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
  sprintf( buf, "%d of type ", argn-1 );
  ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include/stdio.h:188:1: note: 'sprintf' has been explicitly marked deprecated here
/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-math7/lib/ruby/Vector2RUBY_wrap.cxx:1160:3: warning: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")  sprintf( buf, "%d of type ", argn-1 );

^  ^

/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include/stdio.h:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
188:1: note: 'sprintf' has been explicitly marked deprecated here
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-math7/lib/ruby/GaussMarkovProcessRUBY_wrap.cxx:1528:3: warning: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
  sprintf(klass_name, "TYPE%s", type->name);
  ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include/stdio.h:188:1: note: 'sprintf' has been explicitly marked deprecated here
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-math7/lib/ruby/GaussMarkovProcessRUBY_wrap.cxx:1585:5: warning: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
    sprintf(klass_name, "TYPE%s", type->name);
    ^
[... cut ...]
                                                      ^
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
---
Finished <<< gz-math7 [39.5s]

Summary: 1 package finished [39.7s]
  1 package had stderr output: gz-math7
```

Also builds using

```bash
cd ./src/gz-math/build && mkdir build && cd build
cmake .. && make -j8
```

Both builds highlight use of functions deprecated in the version of clang used by Xcode 14.3.1.

mjcarroll, 2023-09-13:
I believe the deprecation warnings come from swig's generated code, they were addressed in upstream, but maybe the version in brew doesn't contain the patch: https://github.com/swig/swig/pull/2548/commits/f94d2fd3d7cafa62d3e771ef52a426fa6be734a2

> The 2 failing tests pass when run independently

The best kind :(

srmainwaring, 2023-09-13:
> maybe the version in brew doesn't contain the patch: [swig/swig@f94d2fd](https://github.com/swig/swig/commit/f94d2fd3d7cafa62d3e771ef52a426fa6be734a2)

Could be my version of swig is outdated. I've avoided doing a full brew upgrade as that will certainly break my ROS 2 Humble from source build (and downstream projects). The osrf/simulation tap is current. I'll recheck when I get round to building the whole shebang again.

closed 2023-09-25 by azeey