gazebosim/gazebo_test_cases #621: gz-msgs: Installation

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/msgs/10/install.html)
- [Source](https://github.com/gazebosim/gz-msgs/tree/gz-msgs10/tutorials/install.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-13:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Details

Continuing from

- https://github.com/gazebosim/gazebo_test_cases/issues/945#issuecomment-1718203548
- https://github.com/gazebosim/gazebo_test_cases/issues/900#issuecomment-1718362032
- https://github.com/gazebosim/gazebo_test_cases/issues/600#issuecomment-1718340446
- https://github.com/gazebosim/gazebo_test_cases/issues/528#issuecomment-1718347066

`gz-msgs10` builds clean

```bash
% colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-select gz-msgs10
Starting >>> gz-msgs10
[Processing: gz-msgs10]                             
[Processing: gz-msgs10]                                     
Finished <<< gz-msgs10 [1min 22s]                              

Summary: 1 package finished [1min 22s]
```

Also builds using

```bash
cd ./src/gz-msgs && mkdir build && cd build
cmake .. && make -j8
```

Test - some failures

```bash
% make test
Running tests...
Test project /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build
      Start  1: INTEGRATION_Factory_TEST
 1/13 Test  #1: INTEGRATION_Factory_TEST .......................Subprocess aborted***Exception:   0.36 sec
      Start  2: check_INTEGRATION_Factory_TEST
 2/13 Test  #2: check_INTEGRATION_Factory_TEST .................   Passed    0.02 sec
      Start  3: INTEGRATION_PointCloudPackedUtils_TEST
 3/13 Test  #3: INTEGRATION_PointCloudPackedUtils_TEST .........Subprocess aborted***Exception:   0.14 sec
      Start  4: check_INTEGRATION_PointCloudPackedUtils_TEST
 4/13 Test  #4: check_INTEGRATION_PointCloudPackedUtils_TEST ...   Passed    0.02 sec
      Start  5: INTEGRATION_Utility_TEST
 5/13 Test  #5: INTEGRATION_Utility_TEST .......................Subprocess aborted***Exception:   0.14 sec
      Start  6: check_INTEGRATION_Utility_TEST
 6/13 Test  #6: check_INTEGRATION_Utility_TEST .................   Passed    0.02 sec
      Start  7: INTEGRATION_gz_TEST
 7/13 Test  #7: INTEGRATION_gz_TEST ............................Subprocess aborted***Exception:   0.14 sec
      Start  8: check_INTEGRATION_gz_TEST
 8/13 Test  #8: check_INTEGRATION_gz_TEST ......................   Passed    0.02 sec
      Start  9: INTEGRATION_headers
 9/13 Test  #9: INTEGRATION_headers ............................Subprocess aborted***Exception:   0.14 sec
      Start 10: check_INTEGRATION_headers
10/13 Test #10: check_INTEGRATION_headers ......................   Passed    0.02 sec
      Start 11: INTEGRATION_image_msg
11/13 Test #11: INTEGRATION_image_msg ..........................Subprocess aborted***Exception:   0.16 sec
      Start 12: check_INTEGRATION_image_msg
12/13 Test #12: check_INTEGRATION_image_msg ....................   Passed    0.02 sec
      Start 13: basic_TEST.py
[... 222 lines cut ...]
[----------] Global test environment set-up.
[----------] 1 test from Messages
[ RUN      ] Messages.NestedHeaders
[       OK ] Messages.NestedHeaders (0 ms)
[----------] 1 test from Messages (0 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (0 ms total)
[  PASSED  ] 1 test.
```

```bash
% ./bin/INTEGRATION_image_msg
Running main() from /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/test/gtest_vendor/src/gtest_main.cc
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from Image
[ RUN      ] Image.PixelFormat
[       OK ] Image.PixelFormat (0 ms)
[----------] 1 test from Image (0 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (0 ms total)
[  PASSED  ] 1 test.
```

mjcarroll, 2023-09-13:
Rather than `make test`, can you use `ctest -V` so we can get the logs?

srmainwaring, 2023-09-13:
```bash
% ctest -V
UpdateCTestConfiguration  from :/Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/DartConfiguration.tcl
Parse Config file:/Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/DartConfiguration.tcl
Parse Config file:/Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/DartConfiguration.tcl
Test project /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
      Start  1: INTEGRATION_Factory_TEST

1: Test command: /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/bin/INTEGRATION_Factory_TEST "--gtest_output=xml:/Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/test_results/INTEGRATION_Factory_TEST.xml"
1: Working Directory: /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/test/integration
1: Test timeout computed to be: 240
1: Running main() from /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/test/gtest_vendor/src/gtest_main.cc
1: [==========] Running 7 tests from 1 test suite.
1: [----------] Global test environment set-up.
1: [----------] 7 tests from FactoryTest
1: [ RUN      ] FactoryTest.Type
1: [       OK ] FactoryTest.Type (0 ms)
1: [ RUN      ] FactoryTest.New
1: [       OK ] FactoryTest.New (0 ms)
1: [ RUN      ] FactoryTest.DeprecatedNonFullyQualified
1: Message (gz.msgs.StringMsg) was retrieved with non-fully qualified name. This behavior is deprecated in msgs10
1: [       OK ] FactoryTest.DeprecatedNonFullyQualified (0 ms)
1: [ RUN      ] FactoryTest.DynamicMessageNoEnv
1: [       OK ] FactoryTest.DynamicMessageNoEnv (0 ms)
1: [ RUN      ] FactoryTest.DynamicMessageWithEnv
1: [       OK ] FactoryTest.DynamicMessageWithEnv (0 ms)
1: [ RUN      ] FactoryTest.NewAllRegisteredTypes
1: [       OK ] FactoryTest.NewAllRegisteredTypes (0 ms)
1: [ RUN      ] FactoryTest.MultipleMessagesInAProto
1: [       OK ] FactoryTest.MultipleMessagesInAProto (0 ms)
1: [----------] 7 tests from FactoryTest (1 ms total)
1: 
1: [----------] Global test environment tear-down
1: [==========] 7 tests from 1 test suite ran. (1 ms total)
1: [  PASSED  ] 7 tests.
 1/13 Test  #1: INTEGRATION_Factory_TEST .......................   Passed    0.04 sec
test 2
      Start  2: check_INTEGRATION_Factory_TEST

2: Test command: /opt/homebrew/Frameworks/Python.framework/Versions/3.11/bin/python3.11 "/Users/rhys/Code/osrf/gz_harmonic_ws/install/share/gz/gz-cmake3/tools/check_test_ran.py" "/Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/test_results/INTEGRATION_Factory_TEST.xml"
2: Working Directory: /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/test/integration
2: Test timeout computed to be: 1500
2: Checking for test results in /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/test_results/INTEGRATION_Factory_TEST.xml
 2/13 Test  #2: check_INTEGRATION_Factory_TEST .................   Passed    0.02 sec
test 3
      Start  3: INTEGRATION_PointCloudPackedUtils_TEST

3: Test command: /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/bin/INTEGRATION_PointCloudPackedUtils_TEST "--gtest_output=xml:/Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/test_results/INTEGRATION_PointCloudPackedUtils_TEST.xml"
3: Working Directory: /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/test/integration
3: Test timeout computed to be: 240
3: Running main() from /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/test/gtest_vendor/src/gtest_main.cc
3: [==========] Running 5 tests from 1 test suite.
3: [----------] Global test environment set-up.
3: [----------] 5 tests from PointCloudPackedUtilsTest
3: [ RUN      ] PointCloudPackedUtilsTest.BadFields
3: Field [x] does not exist.
3: Field [xyz] does not exist.
3: Field [r] does not exist.
3: [       OK ] PointCloudPackedUtilsTest.BadFields (0 ms)
3: [ RUN      ] PointCloudPackedUtilsTest.Operators
3: [       OK ] PointCloudPackedUtilsTest.Operators (0 ms)
3: [ RUN      ] PointCloudPackedUtilsTest.MultipleFields
[... 329 lines cut ...]
13: Hint: make sure your test modules/packages have valid Python names.
13: Traceback:
13: /opt/homebrew/Cellar/python@3.11/3.11.4_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/importlib/__init__.py:126: in import_module
13:     return _bootstrap._gcd_import(name[level:], package, level)
13: ../../python/test/basic_TEST.py:15: in <module>
13:     from gz.msgs10.vector3d_pb2 import Vector3d
13: E   ModuleNotFoundError: No module named 'gz.msgs10'
13: - generated xml file: /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/test_results/basic_TEST.xml -
13: =========================== short test summary info ============================
13: ERROR ../../python/test/basic_TEST.py
13: !!!!!!!!!!!!!!!!!!!! Interrupted: 1 error during collection !!!!!!!!!!!!!!!!!!!!
13: =============================== 1 error in 0.04s ===============================
13/13 Test #13: basic_TEST.py ..................................***Failed    0.17 sec

85% tests passed, 2 tests failed out of 13

Total Test time (real) =   1.09 sec

The following tests FAILED:
	  7 - INTEGRATION_gz_TEST (Failed)
	 13 - basic_TEST.py (Failed)
Errors while running CTest
Output from these tests are in: /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-msgs/build/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
```

mjcarroll, 2023-09-13:
```
libc++abi: terminating due to uncaught exception of type std::__1::__fs::filesystem::filesystem_error: filesystem error: in directory_iterator::directory_iterator(...): No such file or directory ["/usr/local/share/gz/protos"]
```

Hmm, I believe this should make use of the InstallDirectories work from Silvio.  Where is gz-msgs installed in this case?

srmainwaring, 2023-09-13:
> Hmm, I believe this should make use of the InstallDirectories work from Silvio. Where is gz-msgs installed in this case?

No install, tests are run from the build directory in this case.

closed 2023-09-25 by Voldivh