gazebosim/gazebo_test_cases #633: gz-physics: Installation

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/physics/7/installation.html)
- [Source](https://github.com/gazebosim/gz-physics/tree/gz-physics7/tutorials/02_installation.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-13:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### Suggested change

Instructions refer to `ign` branch rather than `gz`:

```diff
- git clone https://github.com/gazebosim/gz-physics -b ign-physics<#>
+ git clone https://github.com/gazebosim/gz-physics -b gz-physics<#>
```

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Details

Continuing from

- https://github.com/gazebosim/gazebo_test_cases/issues/945#issuecomment-1718203548
- https://github.com/gazebosim/gazebo_test_cases/issues/900#issuecomment-1718362032
- https://github.com/gazebosim/gazebo_test_cases/issues/600#issuecomment-1718340446
- https://github.com/gazebosim/gazebo_test_cases/issues/528#issuecomment-1718347066
- https://github.com/gazebosim/gazebo_test_cases/issues/621#issuecomment-1718380277
- https://github.com/gazebosim/gazebo_test_cases/issues/546#issuecomment-1718395524
- https://github.com/gazebosim/gazebo_test_cases/issues/660#issuecomment-1718400408
- https://github.com/gazebosim/gazebo_test_cases/issues/915#issuecomment-1718420051

`gz-physics` builds clean

```bash
% colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-select gz-physics7
Starting >>> gz-physics7
[Processing: gz-physics7]                             
Finished <<< gz-physics7 [43.3s]                       

Summary: 1 package finished [43.8s]
```

Also builds using

```bash
cd ./src/gz-physics && mkdir build && cd build
cmake .. && make -j8
```

Tests - a number of tests are failing.

Voldivh, 2023-09-14:
@azeey we are having some test failures. Are these known issues?

azeey, 2023-09-14:
`COMMON_TEST_joint_features_bullet-featherstone` might be fixed by https://github.com/gazebosim/gz-physics/pull/539 . The other two appear to be tolerance issues. Those seem to crop up with ARM architectures and since we don't have any ARM machines on CI, we don't catch them.

azeey, 2023-09-20:
Closing this since https://github.com/gazebosim/gz-physics/pull/539 is merged.

closed 2023-09-20 by azeey