gazebosim/gazebo_test_cases #669: gz-rendering: Actor animation tutorial

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/rendering/8/actor_animation.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/tutorials/10_actor_animation_tutorial.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

Summary:

- Both render engines are working.
- There is a segmentation fault on exit (`ogre` and `ogre2 metal`).
- The text is not clear (a fainter second displaced image of the text appears under the first - not captured in the screenshot). 

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Test

`ogre` render engine

[image]

`ogre2 metal` render engine

[image]

Excerpt from Apple crash report:

```bash
Time Since Wake:       23497 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000001188dba08
Exception Codes:       0x0000000000000001, 0x00000001188dba08

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [28778]

VM Region Info: 0x1188dba08 is not in any region.  Bytes after previous region: 2210313  Bytes before following region: 2295288
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      __LINKEDIT                  1186a4000-1186c0000    [  112K] r--/rwx SM=COW  ...X.2.3.1.dylib
--->  GAP OF 0x44c000 BYTES
      __TEXT                      118b0c000-118b18000    [   48K] r-x/rwx SM=COW  ...polines.dylib

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   actor_animation               	       0x102940724 std::__1::__shared_count::__release_shared[abi:v15006]() + 56
1   actor_animation               	       0x1029406cc std::__1::__shared_weak_count::__release_shared[abi:v15006]() + 28
2   actor_animation               	       0x10294d060 std::__1::shared_ptr<gz::rendering::v8::RayQuery>::~shared_ptr[abi:v15006]() + 64
3   actor_animation               	       0x1029487e8 std::__1::shared_ptr<gz::rendering::v8::RayQuery>::~shared_ptr[abi:v15006]() + 28
4   libsystem_c.dylib             	       0x188651ec4 __cxa_finalize_ranges + 476
5   libsystem_c.dylib             	       0x188651c4c exit + 44
6   actor_animation               	       0x10294bfdc keyboardCB(unsigned char, int, int) + 116
7   GLUT                          	       0x1c97e6cc8 -[GLUTView keyDown:] + 292
8   AppKit                        	       0x18bbbef7c -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 480
9   AppKit                        	       0x18bbbebc8 -[NSWindow(NSEventRouting) sendEvent:] + 284
10  GLUT                          	       0x1c97e30d4 -[GLUTWindow sendEvent:] + 44
11  AppKit                        	       0x18bbbdf0c -[NSApplication(NSEvent) sendEvent:] + 1556
12  GLUT                          	       0x1c97ed1c4 -[GLUTApplication _runMainLoopUntilDate:autoreleasePool:] + 88
13  GLUT                          	       0x1c97ed2d0 -[GLUTApplication run] + 180
14  GLUT                          	       0x1c97f9958 glutMainLoop + 192
15  actor_animation               	       0x10294c7e4 run(std::__1::vector<std::__1::shared_ptr<gz::rendering::v8::Camera>, std::__1::allocator<std::__1::shared_ptr<gz::rendering::v8::Camera>>>, std::__1::vector<std::__1::shared_ptr<gz::rendering::v8::Visual>, std::__1::allocator<std::__1::shared_ptr<gz::rendering::v8::Visual>>> const&, std::__1::shared_ptr<gz::common::Skeleton>) + 300
16  actor_animation               	       0x10293ed40 main + 1004
17  dyld                          	       0x18843bf28 start + 2236
```

Voldivh, 2023-09-15:
Thanks @srmainwaring , I'll be closing this issue since instructions work.

closed 2023-09-15 by Voldivh