gazebosim/gazebo_test_cases #672: gz-rendering: Bounding Box camera

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/rendering/8/boundingbox_camera.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/tutorials/24_boundingbox_camera_tutorial.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [ ] Instructions work
- [ ] Images (if there are any) match the result

- All examples are failing.

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Test

First example not working:

```bash
% ./boundingbox_camera
[Msg] Loading plugin [gz-rendering-ogre2]
UNSUPPORTED (log once): gldCopyBufferSubData: NEEDS IMPLEMENTATION
libc++abi: terminating due to uncaught exception of type Ogre::RenderingAPIException: OGRE EXCEPTION(3:RenderingAPIException): Fragment Program Ogre/DPSM/CubeToDpsm_ps_GLSL failed to compile. See compile log above for details. in GLSLShader::compile at /tmp/ogre2.3-20230526-26210-1dj9624/ogre-next-2.3.1/RenderSystems/GL3Plus/src/GLSL/OgreGLSLShader.cpp (line 364)
zsh: abort      ./boundingbox_camera
```

The other examples also fail with similar errors.

```bash
% ./boundingbox_camera 2D_visible
```

```bash
% ./boundingbox_camera 2D_full
```

azeey, 2023-09-20:
I've ticketed an [issue](https://github.com/gazebosim/gz-rendering/issues/914). This is not likely to be fixed for Harmonic, so I'll go ahead and close this issue now.