gazebosim/gazebo_test_cases #675: gz-rendering: Camera tracking

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/rendering/8/camera_tracking.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/tutorials/14_camera_tracking_tutorial.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Test

- Both render engine versions working.
- Camera tracking with key and mouse response as expected.
- Segmentation fault on exit - similar to issues reported on other `gz-rendering` examples.

`ogre` render engine

[image]

`ogre2` metal render engine

[image]

Voldivh, 2023-09-14:
Thanks @srmainwaring , we are looking into the issue with the segfault at exit. I'll be closing this issue as completed.

closed 2023-09-14 by Voldivh