gazebosim/gazebo_test_cases #684: gz-rendering: Custom shaders

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/rendering/8/custom_shaders.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/tutorials/15_custom_shaders_tutorial.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Test

Both render engines working correctly.

`ogre` render engine

[image]
[image]

`ogre2 metal` render engine

[image]
[image]

Voldivh, 2023-09-14:
Thanks @srmainwaring

closed 2023-09-14 by Voldivh