gazebosim/gazebo_test_cases #690: gz-rendering: Environment Map

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/rendering/8/envmap.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/tutorials/22_environment_map.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Test

Tested the `ogre2_demo`, did not test creating of skyboxes for the environment map.

Working as expected for `ogre2 metal`. Same segmentation fault on exit as previously reported.

[image]

Voldivh, 2023-09-14:
Thanks @srmainwaring

closed 2023-09-14 by Voldivh