gazebosim/gazebo_test_cases #693: gz-rendering: Gazebo scene viewer

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/rendering/8/gazebo_scene_viewer.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/tutorials/11_gazebo_scene_viewer_tutorial.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
Could not test on a M1 mac - issues installing the dependency `gazebo11` using brew. Looks like a known issue.

```bash
% brew install gazebo11
...
cd /tmp/gazebo11-20230914-20688-6k9pob/gazebo-11.13.0/build/gazebo && /opt/homebrew/Cellar/cmake/3.27.4/bin/cmake -E cmake_symlink_library libgazebo.11.13.0.dylib libgazebo.11.dylib libgazebo.dylib
[ 73%] Built target libgazebo
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
Failed to build gazebo11 on Apple Silicon https://github.com/osrf/homebrew-simulation/issues/2220
```

azeey, 2023-09-20:
wow, I didn't know we had a tutorial that depends on `gazebo11`. @iche033 should we remove it?

iche033, 2023-09-20:
yes I vote to remove it as well. Gazebo-classic pulls in different gz library versions and that causes issues at run time. I'll remove the example.

Voldivh, 2023-09-25:
I'll be closing this since the PR have been merged.