gazebosim/gazebo_test_cases #696: gz-rendering: Heightmap

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/rendering/8/heightmap.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/tutorials/21_heightmap.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Test

The `ogre` render engine version is failing:

```bash
% ./heightmap 
[Msg] Loading plugin [gz-rendering-ogre2]
UNSUPPORTED (log once): gldCopyBufferSubData: NEEDS IMPLEMENTATION
[Wrn] [Ogre2Heightmap.cc:125] Heightmap final sampling should be 2^n
 which differs from ogre1's 2^n+1
The last row and column will be cropped
size = (width * sampling) - sampling + 1
[257] = ([129] * [2]) - [2] + 1
Error starting up: ogre2
[Err] [GlutWindow.cc:320] No cameras found. Scene will not be rendered
```

The `ogre2 metal` render engine version runs with some warnings:

```bash
% ./heightmap ogre2 metal
[Msg] Loading plugin [gz-rendering-ogre2]
[Wrn] [Ogre2Heightmap.cc:125] Heightmap final sampling should be 2^n
 which differs from ogre1's 2^n+1
The last row and column will be cropped
size = (width * sampling) - sampling + 1
[257] = ([129] * [2]) - [2] + 1
[Msg] Loading heightmap: example_bowl
[Msg] Heightmap loaded. Process took 0 ms.
[Wrn] [Ogre2Heightmap.cc:125] Heightmap final sampling should be 2^n
 which differs from ogre1's 2^n+1
The last row and column will be cropped
size = (width * sampling) - sampling + 1
[513] = ([257] * [2]) - [2] + 1
[Msg] Loading heightmap: example_city
[Msg] Heightmap loaded. Process took 0 ms.
Invalid parameter at line 418 (@foreach). 'first_valid_detail_map_nm' is not a number nor a variable
===============================
  TAB - Switch render engines  
  ESC - Exit                   
===============================
```

The heighmap scaling seem to be too large?

[image]

azeey, 2023-09-20:
> The heighmap scaling seem to be too large?

@iche033 is this a known issue?

iche033, 2023-09-20:
yeah we could probably reduce z scale to make it more realistic but on the other hand its exaggerated in z a little to showcase the blending of textures at different heights. I think it may be fine for now. One thing we could do is add a note in the example code or tutorial to mention that the heightmap is not to scale

azeey, 2023-09-20:
Okay, but it seems like it's not a macOS issue. I'll go ahead and close this ticket then.

closed 2023-09-20 by azeey