gazebosim/gazebo_test_cases #699: gz-rendering: Installation

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/rendering/8/installation.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/tutorials/02_install.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### Suggested changes

Update the branch details in the instructions

```diff
- git clone https://github.com/gazebosim/gz-rendering -b ign-rendering<#>
+ git clone https://github.com/gazebosim/gz-rendering -b gz-rendering<#>
```

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Details

Continuing from

- https://github.com/gazebosim/gazebo_test_cases/issues/945#issuecomment-1718203548
- https://github.com/gazebosim/gazebo_test_cases/issues/900#issuecomment-1718362032
- https://github.com/gazebosim/gazebo_test_cases/issues/600#issuecomment-1718340446
- https://github.com/gazebosim/gazebo_test_cases/issues/528#issuecomment-1718347066
- https://github.com/gazebosim/gazebo_test_cases/issues/621#issuecomment-1718380277
- https://github.com/gazebosim/gazebo_test_cases/issues/546#issuecomment-1718395524
- https://github.com/gazebosim/gazebo_test_cases/issues/660#issuecomment-1718400408
- https://github.com/gazebosim/gazebo_test_cases/issues/915#issuecomment-1718420051
- https://github.com/gazebosim/gazebo_test_cases/issues/633#issuecomment-1718454984

`gz-rendering` builds

```bash
 % colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-select gz-rendering8
Starting >>> gz-rendering8
[Processing: gz-rendering8]                             
[Processing: gz-rendering8]                                 
--- stderr: gz-rendering8                                          
CMake Warning (dev) at CMakeLists.txt:174 (find_package):
  Policy CMP0146 is not set: The FindCUDA module is removed.  Run "cmake
  --help-policy CMP0146" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CUDA_TOOLKIT_ROOT_DIR not found or specified
CMake Warning at /Users/rhys/Code/osrf/gz_harmonic_ws/install/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:68 (message):
   CONFIGURATION WARNINGS:
   -- Skipping component [optix]: Missing dependency [OptiX].
      ^~~~~ Set SKIP_optix=true in cmake to suppress this warning.
   
Call Stack (most recent call first):
  CMakeLists.txt:216 (gz_configure_build)

---
Finished <<< gz-rendering8 [1min 16s]

Summary: 1 package finished [1min 16s]
  1 package had stderr output: gz-rendering8
```

Also builds using

```bash
cd ./src/gz-rendering && mkdir build && cd build
cmake .. && make -j8
[... 587 lines cut ...]
[----------] 3 tests from RenderingIfaceTest
[ RUN      ] RenderingIfaceTest.HasEngine
[       OK ] RenderingIfaceTest.HasEngine (0 ms)
[ RUN      ] RenderingIfaceTest.GetEngine
[Dbg] [RenderingIface.cc:135] No rendering engine is loaded yet
[Msg] Loading plugin []
Error while loading the library [/Users/rhys/Code/osrf/gz_harmonic_ws/install/lib/]: dlopen(/Users/rhys/Code/osrf/gz_harmonic_ws/install/lib/, 0x0005): tried: '/Users/rhys/Code/osrf/gz_harmonic_ws/install/lib/' (not a file), '/Users/rhys/Code/osrf/gz_harmonic_ws/install/lib/' (not a file), '/System/Volumes/Preboot/Cryptexes/OS/Users/rhys/Code/osrf/gz_harmonic_ws/install/lib/' (no such file), '/Users/rhys/Code/osrf/gz_harmonic_ws/install/lib/' (not a file)
[Err] [RenderEngineManager.cc:494] Failed to load plugin [] : couldn't load library on path [/Users/rhys/Code/osrf/gz_harmonic_ws/install/lib/].
/Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-rendering/test/common_test/RenderingIface_TEST.cc:88: Failure
Expected: (nullptr) != (eng), actual: (nullptr) vs NULL
[  FAILED  ] RenderingIfaceTest.GetEngine (0 ms)
[ RUN      ] RenderingIfaceTest.RegisterEngine
[Msg] Loading plugin [gz-rendering-ogre]
[Err] [RenderEngineManager.cc:340] Invalid render-engine index: 1000000
[       OK ] RenderingIfaceTest.RegisterEngine (84 ms)
[----------] 3 tests from RenderingIfaceTest (84 ms total)

[----------] Global test environment tear-down
[==========] 3 tests from 1 test suite ran. (85 ms total)
[  PASSED  ] 2 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] RenderingIfaceTest.GetEngine

 1 FAILED TEST
```

Voldivh, 2023-09-15:
Suggested changes have been merged in https://github.com/gazebosim/gz-rendering/pull/906. Thanks @srmainwaring

azeey, 2023-09-25:
https://github.com/gazebosim/gz-rendering/issues/919

closed 2023-09-25 by azeey