gazebosim/gazebo_test_cases #705: gz-rendering: Mesh Viewer

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/rendering/8/mesh_viewer.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/tutorials/12_mesh_viewer_tutorial.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [ ] Instructions work
- [ ] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Test

`ogre` version runs, rendering does not match picture, segmentation fault on exit.
 
```
./mesh_viewer 
[Msg] Loading plugin [gz-rendering-ogre]
===============================
  TAB - Switch render engines  
  ESC - Exit                   
===============================
zsh: segmentation fault  ./mesh_viewer
```

[image]

`ogre2 metal` version runs, rendering does not match picture, segmentation fault on exit.

```bash
./mesh_viewer ogre2 metal
[Msg] Loading plugin [gz-rendering-ogre2]
===============================
  TAB - Switch render engines  
  ESC - Exit                   
===============================
zsh: segmentation fault  ./mesh_viewer ogre2 metal
```

[image]

Apple crash report for `ogre` render engine

<details>

```console
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               mesh_viewer [78012]
Path:                  /Users/USER/*/mesh_viewer
Identifier:            mesh_viewer
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        zsh [21506]
Responsible:           Terminal [51221]
User ID:               501

Date/Time:             2023-09-14 12:21:30.5977 +0100
OS Version:            macOS 13.4 (22F66)
Report Version:        12
Anonymous UUID:        F22C2801-4D85-DE28-E2BB-F2C77744711E

Sleep/Wake UUID:       E3A6B165-F4F5-4A64-99BE-584151551A70

Time Awake Since Boot: 780000 seconds
Time Since Wake:       15036 seconds

System Integrity Protection: enabled
[... 3260 lines cut ...]
    "name" : "libgif.7.2.0.dylib"
  },
  {
    "source" : "P",
    "arch" : "arm64",
    "base" : 4434837504,
    "size" : 950272,
    "uuid" : "601ca77a-6f65-3adb-9750-293e944e1288",
    "path" : "\/opt\/homebrew\/*\/libnetcdf.19.dylib",
    "name" : "libnetcdf.19.dylib"
  },
  {
    "source" : "P",
    "arch" : "arm64",
    "base" : 4441030656,
    "size" : 1015808,
    "uuid" : "e6e461e7-3485-33d3-8cec-04863cb02505",
    "path" : "\/opt\/homebrew\/*\/libcfitsio.10.4.2.0.dylib",
    "name" : "libcfitsio.10.4.2.0.dylib"
  },
  {
    "source" : "P",
    "arch" : "arm64",
    "base" : 4447371264,
    "size" : 306

azeey, 2023-09-25:
https://github.com/gazebosim/gz-rendering/issues/917

closed 2023-09-25 by azeey