gazebosim/gazebo_test_cases #708: gz-rendering: Particles

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/rendering/8/particles.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/tutorials/20_particles_tutorial.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Test

ogre version does not work

```bash
./particles_demo 
[Msg] Loading plugin [gz-rendering-ogre2]
UNSUPPORTED (log once): gldCopyBufferSubData: NEEDS IMPLEMENTATION
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
libc++abi: terminating due to uncaught exception of type Ogre::RenderingAPIException: OGRE EXCEPTION(3:RenderingAPIException): Fragment Program 100000001PixelShader_ps failed to compile. See compile log above for details. in GLSLShader::compile at /tmp/ogre2.3-20230526-26210-1dj9624/ogre-next-2.3.1/RenderSystems/GL3Plus/src/GLSL/OgreGLSLShader.cpp (line 364)
zsh: abort      ./particles_demo
```

ogre2 version runs but reports segmentation fault on exit

```bash
./particles_demo ogre2 metal
[Msg] Loading plugin [gz-rendering-ogre2]
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
===============================
  TAB - Switch render engines  
  ESC - Exit                   
===============================
2023-09-14 12:12:32.993 particles_demo[77830:36450509] DEBUG Pre: Retain count is 3
2023-09-14 12:12:32.993 particles_demo[77830:36450509] DEBUG Pos: Retain count is 4
2023-09-14 12:12:32.994 particles_demo[77830:36450509] DEBUG Pos2: Retain count is 3
2023-09-14 12:12:32.994 particles_demo[77830:36450509] DEBUG Pre: Retain count is 2
2023-09-14 12:12:32.994 particles_demo[77830:36450509] DEBUG Pos: Retain count is 3
2023-09-14 12:12:32.995 particles_demo[77830:36450509] DEBUG Pos2: Retain count is 2
2023-09-14 12:12:32.995 particles_demo[77830:36450509] DEBUG Pre: Retain count is 1
2023-09-14 12:12:32.995 particles_demo[77830:36450509] DEBUG Pos: Retain count is 2
2023-09-14 12:12:32.996 particles_demo[77830:36450509] DEBUG Pos2: Retain count is 1
zsh: segmentation fault  ./particles_demo ogre2 metal
rhys@MacBookPro2 build % ./particles_demo ogre2 metal
[Msg] Loading plugin [gz-rendering-ogre2]
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
[Dbg] [Ogre2ParticleEmitter.cc:504] Particle emitter initialized
===============================
  TAB - Switch render engines  
  ESC - Exit                   
===============================
2023-09-14 12:12:46.362 particles_demo[77836:36450748] DEBUG Pre: Retain count is 3
2023-09-14 12:12:46.362 particles_demo[77836:36450748] DEBUG Pos: Retain count is 4
2023-09-14 12:12:46.362 particles_demo[77836:36450748] DEBUG Pos2: Retain count is 3
2023-09-14 12:12:46.362 particles_demo[77836:36450748] DEBUG Pre: Retain count is 2
2023-09-14 12:12:46.362 particles_demo[77836:36450748] DEBUG Pos: Retain count is 3
2023-09-14 12:12:46.363 particles_demo[77836:36450748] DEBUG Pos2: Retain count is 2
2023-09-14 12:12:46.363 particles_demo[77836:36450748] DEBUG Pre: Retain count is 1
[... 2382 lines cut ...]
      "deploymentId" : 400000019
    },
    {
      "treatmentId" : "3a3cf641-8471-4e4e-9ad4-81d0ede970fd",
      "experimentId" : "64a84dae90d82611a0bd7d3d",
      "deploymentId" : 400000004
    }
  ]
}
}

Model: MacBookPro18,2, BootROM 8422.121.1, proc 10:8:2 processors, 32 GB, SMC 
Graphics: Apple M1 Max, Apple M1 Max, Built-In
Display: Color LCD, 3456 x 2234 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Samsung
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Apr  7 2023 15:18:26 version 20.96.28.1.8.7.146 FWID 01-aff1b48a
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

azeey, 2023-09-25:
https://github.com/gazebosim/gz-rendering/issues/918

closed 2023-09-25 by azeey