gazebosim/gazebo_test_cases #711: gz-rendering: README

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/libs/rendering)
- [Source](https://github.com/gazebosim/gz-rendering/blob/gz-rendering8/README.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Suggested changes

- Installation tutorial (2 places) links to an old version of the rendering library: https://gazebosim.org/api/rendering/5.0/installation.html
- Change to: https://gazebosim.org/api/rendering/8/installation.html

iche033, 2023-09-14:
suggested changes done in:
https://github.com/gazebosim/gz-rendering/pull/906
https://github.com/gazebosim/gz-rendering/pull/907

Voldivh, 2023-09-15:
Thanks @srmainwaring and @iche033 , I'll be closing this since both PR with changes have been merged.

closed 2023-09-15 by Voldivh