gazebosim/gazebo_test_cases #720: gz-rendering: Simple demo

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/rendering/8/simple_demo.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/tutorials/18_simple_demo_tutorial.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
```bash
./simple_demo
```

[image]

```bash
./simple_demo ogre2 metal
```

[image]

Voldivh, 2023-09-14:
Thanks @srmainwaring !