gazebosim/gazebo_test_cases #723: gz-rendering: Text geometry

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/rendering/8/text_geom.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/tutorials/19_text_geom_tutorial.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Test

```bash
./text_geom 
[Msg] Loading plugin [gz-rendering-ogre]
===============================
  TAB - Switch render engines  
  ESC - Exit                   
===============================
```

[image]

```bash
rhys@MacBookPro2 build % ./text_geom ogre2 metal
[Msg] Loading plugin [gz-rendering-ogre]
===============================
  TAB - Switch render engines  
  ESC - Exit                   
===============================
```

[image]

Voldivh, 2023-09-14:
Thanks @srmainwaring

closed 2023-09-14 by Voldivh