gazebosim/gazebo_test_cases #729: gz-rendering: Understanding the Rendering Plugin

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/rendering/8/renderingplugin.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/tutorials/03_rendering_plugins.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Test

```bash
export GZ_SIM_RENDER_ENGINE_PATH=$HOME/Code/osrf/gz_harmonic_ws/src/gz-rendering/examples/hello_world_plugin/build
gz sim -v4 -s -r --render-engine HelloWorldPlugin empty.sdf
```

```bash
export GZ_SIM_RENDER_ENGINE_PATH=$HOME/Code/osrf/gz_harmonic_ws/src/gz-rendering/examples/hello_world_plugin/build
gz sim -v4 -g --render-engine HelloWorldPlugin
```

[image]

Voldivh, 2023-09-14:
Thanks @srmainwaring

closed 2023-09-14 by Voldivh