gazebosim/gazebo_test_cases #735: gz-sensors: Installation

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/sensors/8/installation.html)
- [Source](https://github.com/gazebosim/gz-sensors/tree/gz-sensors8/tutorials/install.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Details

Continuing from

- https://github.com/gazebosim/gazebo_test_cases/issues/564#issuecomment-1718901080

`gz-sensors` builds clean

```bash
colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-select gz-sensors8
Starting >>> gz-sensors8
Finished <<< gz-sensors8 [15.1s]                       

Summary: 1 package finished [15.6s]
```

Also builds from

```bash
cd ./src/gz-sensors && mkdir build && cd build
cmake .. && make -j8
```

Test - clear

```bash
% make test
Running tests...
Test project /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-sensors/build
      Start  1: UNIT_Manager_TEST
 1/54 Test  #1: UNIT_Manager_TEST ................................   Passed    0.69 sec
      Start  2: check_UNIT_Manager_TEST
 2/54 Test  #2: check_UNIT_Manager_TEST ..........................   Passed    0.02 sec
      Start  3: UNIT_Noise_TEST
 3/54 Test  #3: UNIT_Noise_TEST ..................................   Passed    0.25 sec
      Start  4: check_UNIT_Noise_TEST
 4/54 Test  #4: check_UNIT_Noise_TEST ............................   Passed    0.02 sec
      Start  5: UNIT_Sensor_TEST
 5/54 Test  #5: UNIT_Sensor_TEST .................................   Passed    2.49 sec
      Start  6: check_UNIT_Sensor_TEST
 6/54 Test  #6: check_UNIT_Sensor_TEST ...........................   Passed    0.03 sec
      Start  7: UNIT_Util_TEST
 7/54 Test  #7: UNIT_Util_TEST ...................................   Passed    0.28 sec
      Start  8: check_UNIT_Util_TEST
 8/54 Test  #8: check_UNIT_Util_TEST .............................   Passed    0.02 sec
      Start  9: UNIT_Lidar_TEST
 9/54 Test  #9: UNIT_Lidar_TEST ..................................   Passed    0.41 sec
      Start 10: check_UNIT_Lidar_TEST
10/54 Test #10: check_UNIT_Lidar_TEST ............................   Passed    0.02 sec
      Start 11: UNIT_Camera_TEST
11/54 Test #11: UNIT_Camera_TEST .................................   Passed    0.51 sec
      Start 12: check_UNIT_Camera_TEST
12/54 Test #12: check_UNIT_Camera_TEST ...........................   Passed    0.02 sec
      Start 13: UNIT_ImuSensor_TEST
13/54 Test #13: UNIT_ImuSensor_TEST ..............................   Passed    0.55 sec
      Start 14: check_UNIT_ImuSensor_TEST
14/54 Test #14: check_UNIT_ImuSensor_TEST ........................   Passed    0.02 sec
      Start 15: INTEGRATION_boundingbox_camera
15/54 Test #15: INTEGRATION_boundingbox_camera ...................   Passed    1.27 sec
[... 58 lines cut ...]
      Start 45: INTEGRATION_force_torque
45/54 Test #45: INTEGRATION_force_torque .........................   Passed    0.35 sec
      Start 46: check_INTEGRATION_force_torque
46/54 Test #46: check_INTEGRATION_force_torque ...................   Passed    0.02 sec
      Start 47: INTEGRATION_imu
47/54 Test #47: INTEGRATION_imu ..................................   Passed    0.30 sec
      Start 48: check_INTEGRATION_imu
48/54 Test #48: check_INTEGRATION_imu ............................   Passed    0.02 sec
      Start 49: INTEGRATION_logical_camera
49/54 Test #49: INTEGRATION_logical_camera .......................   Passed    0.31 sec
      Start 50: check_INTEGRATION_logical_camera
50/54 Test #50: check_INTEGRATION_logical_camera .................   Passed    0.02 sec
      Start 51: INTEGRATION_magnetometer
51/54 Test #51: INTEGRATION_magnetometer .........................   Passed    0.35 sec
      Start 52: check_INTEGRATION_magnetometer
52/54 Test #52: check_INTEGRATION_magnetometer ...................   Passed    0.02 sec
      Start 53: INTEGRATION_navsat
53/54 Test #53: INTEGRATION_navsat ...............................   Passed    0.34 sec
      Start 54: check_INTEGRATION_navsat
54/54 Test #54: check_INTEGRATION_navsat .........................   Passed    0.02 sec

100% tests passed, 0 tests failed out of 54

Total Test time (real) =  28.43 sec
```

Voldivh, 2023-09-14:
Thanks @srmainwaring!

closed 2023-09-14 by Voldivh