gazebosim/gazebo_test_cases #750: gz-sim: Apply Force and Torque

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/sim/8/apply_force_torque.html)
- [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim8/tutorials/apply_force_torque.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Suggested changes

- Images in the tutorial are too large, and render wider than a standard browser window.
- In Chrome you cannot scroll sideways to see the full image.
- On a laptop the screen is not wide enough to see the images even in full screen mode.

### Testing

- [x] Apply force to link
- [x] Apply torque to a link
- [x] Apply force with an offset
- [x] Rotation tool
- [x] Apply All

[image]

Voldivh, 2023-09-14:
Thanks @srmainwaring, I'll take a look into the images being too large.

azeey, 2023-09-25:
https://github.com/gazebosim/gz-sim/pull/2169

closed 2023-09-25 by azeey