gazebosim/gazebo_test_cases #795: gz-sim: GUI Configuration

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/sim/8/gui_config.html)
- [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim8/tutorials/gui_config.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Testing

- [x] Default configuration.
- [x] Change to dark theme.

[image]

- [x] Restore to default.
- [x] SDF override.

[image]
[image]

- [x] Command line.

[image]

The command line for macOS is slightly different as the gui is started separately from the server; for example:

```bash
 gz sim -v4 -g --gui-config fuel_preview.config
```

Voldivh, 2023-09-14:
Thanks @srmainwaring maybe we could add a little note with the clarification of the command line wdyt?

iche033, 2023-09-14:
related note: this is mentioned in the [Get Started](https://gazebosim.org/docs/harmonic/getstarted#macos) guide for macOS users.

srmainwaring, 2023-09-14:
There is also a helpful message if you try to run the combined server / gui combo on macOS:

```bash
gz sim -v4 -r shapes.sdf 
On macOS `gz sim` currently only works with either the -s argument
or the -g argument, you cannot run both server and gui in one terminal.
See https://github.com/gazebosim/gz-sim/issues/44 for more info.
```

So it's all good.

Voldivh, 2023-09-14:
Oh ok I wasn't aware. I'll be closing this as completed then. Thanks @srmainwaring and @iche033

closed 2023-09-14 by Voldivh