gazebosim/gazebo_test_cases #804: gz-sim: Installation

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/sim/8/install.html)
- [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim8/tutorials/install.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Details

Continuing from

- https://github.com/gazebosim/gazebo_test_cases/issues/735#issuecomment-1718915088

`gz-sim` builds

```bash
% colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-select gz-sim8    
Starting >>> gz-sim8 
[Processing: gz-sim8]                             
[Processing: gz-sim8]                                 
[Processing: gz-sim8]                                       
[Processing: gz-sim8]                                       
[Processing: gz-sim8]                                       
[Processing: gz-sim8]                                       
[Processing: gz-sim8]                                       
[Processing: gz-sim8]                                       
[Processing: gz-sim8]                                       
[Processing: gz-sim8]                                       
[Processing: gz-sim8]                                       
[Processing: gz-sim8]                                       
[Processing: gz-sim8]                                       
--- stderr: gz-sim8                                          
CMake Warning (dev) at /opt/homebrew/lib/cmake/protobuf/protobuf-options.cmake:6 (option):
  Policy CMP0077 is not set: option() honors normal variables.  Run "cmake
  --help-policy CMP0077" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  For compatibility with older versions of CMake, option is clearing the
  normal variable 'protobuf_MODULE_COMPATIBLE'.
Call Stack (most recent call first):
  /opt/homebrew/lib/cmake/protobuf/protobuf-config.cmake:2 (include)
  /Users/rhys/Code/osrf/gz_harmonic_ws/install/share/cmake/gz-cmake3/cmake3/FindGzProtobuf.cmake:29 (find_package)
  /Users/rhys/Code/osrf/gz_harmonic_ws/install/share/cmake/gz-cmake3/cmake3/GzFindPackage.cmake:243 (find_package)
  CMakeLists.txt:198 (gz_find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< gz-sim8 [6min 42s]

Summary: 1 package finished [6min 42s]
```

Also build using

```bash
cd ./src/gz-sim && mkdir build && cd build
cmake .. && make -j8
```

Test - some tests fail

```bash
% make test
Running tests...
Test project /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-sim/build
        Start   1: UNIT_Actor_TEST
[... 644 lines cut ...]
The following tests FAILED:
	 37 - UNIT_Server_TEST (Failed)
	 49 - UNIT_Util_TEST (Failed)
	 67 - UNIT_ModelCommandAPI_TEST (Failed)
	 89 - INTEGRATION_acoustic_comms (Failed)
	103 - INTEGRATION_battery_plugin (Failed)
	205 - INTEGRATION_physics_system (Failed)
	223 - INTEGRATION_scene_broadcaster_system (Failed)
	225 - INTEGRATION_sdf_frame_semantics (Failed)
	241 - INTEGRATION_user_commands (SEGFAULT)
	253 - INTEGRATION_world_control_state (Failed)
	305 - INTEGRATION_python_system_loader (Failed)
	313 - actor_TEST (Failed)
	314 - joint_TEST (Failed)
	315 - light_TEST (Failed)
	316 - link_TEST (Failed)
	317 - model_TEST (Failed)
	318 - sensor_TEST (Failed)
	319 - testFixture_TEST (Failed)
	320 - world_TEST (Failed)
Errors while running CTest
Output from these tests are in: /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-sim/build/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
make: *** [test] Error 8
```

Voldivh, 2023-09-14:
@azeey we are having some test failures according to the output shown by @srmainwaring , are these known issues?

azeey, 2023-09-14:
Our [macOS CI](https://build.osrfoundation.org/view/ign-harmonic/job/ignition_gazebo-ci-gz-sim8-homebrew-amd64/) is green, but it runs on BigSur and Monterey on Intel machines, so I would say these are not know issues. However, fixing them will be difficult, again, since we don't have CI for M1 machines yet.

closed 2023-09-25 by azeey