gazebosim/gazebo_test_cases #852: gz-sim: Model Photo Shoot

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/sim/8/model_photo_shoot.html)
- [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim8/tutorials/model_photo_shoot.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Test

Example works as expected. At one point this was not working in `gz-garden`, so great to see it operational.

[image]
[image]
[image]
[image]
[image]

Voldivh, 2023-09-14:
Thanks @srmainwaring

closed 2023-09-14 by Voldivh