gazebosim/gazebo_test_cases #873: gz-sim: Rendering plugins

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/sim/8/rendering_plugins.html)
- [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim8/tutorials/rendering_plugins.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-14:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Test

```bash
% export GZ_SIM_SYSTEM_PLUGIN_PATH=`pwd`/build
% gz sim -v4 -s -r rendering_plugins.sdf
```

```bash
% export GZ_GUI_PLUGIN_PATH=`pwd`/build
% gz sim -v4 -g
```

[image]

Voldivh, 2023-09-14:
Thanks @srmainwaring

closed 2023-09-14 by Voldivh