gazebosim/gazebo_test_cases #915: gz-transport: Installation

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/transport/13/installation.html)
- [Source](https://github.com/gazebosim/gz-transport/tree/gz-transport13/tutorials/02_installation.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-13:
### Status

- [x] Instructions work
- [x] Images (if there are any) match the result

### System

- MacBook Pro 2021, Apple M1 Max, 32 GB
- macOS Ventura 13.4
- Xcode 14.3.1 

### Details

Continuing from

- https://github.com/gazebosim/gazebo_test_cases/issues/945#issuecomment-1718203548
- https://github.com/gazebosim/gazebo_test_cases/issues/900#issuecomment-1718362032
- https://github.com/gazebosim/gazebo_test_cases/issues/600#issuecomment-1718340446
- https://github.com/gazebosim/gazebo_test_cases/issues/528#issuecomment-1718347066
- https://github.com/gazebosim/gazebo_test_cases/issues/621#issuecomment-1718380277
- https://github.com/gazebosim/gazebo_test_cases/issues/546#issuecomment-1718395524
- https://github.com/gazebosim/gazebo_test_cases/issues/660#issuecomment-1718400408

`gz-transport13` builds with a warning

```bash
% colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-select gz-transport13
Starting >>> gz-transport13
--- stderr: gz-transport13                                
CMake Warning (dev) at /Users/rhys/Code/osrf/gz_harmonic_ws/install/share/cmake/gz-cmake3/cmake3/GzFindPackage.cmake:456 (message):
   -- THIS MESSAGE IS INTENDED FOR GZ-TRANSPORT AUTHORS --
      (IF YOU SEE THIS, PLEASE REPORT IT)

  Could not find pkg-config information for UUID.  It was not provided by the
  find-module for the package, nor was it explicitly passed into the call to
  gz_find_package(~).  This is most likely an error in this project's use of
  gz-cmake.
Call Stack (most recent call first):
  CMakeLists.txt:94 (gz_find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< gz-transport13 [14.3s]

Summary: 1 package finished [14.7s]
  1 package had stderr output: gz-transport13
```

Also builds using

```bash
cd ./src/gz-transport && mkdir build && cd build
cmake .. && make -j8
```

Test (must accept incoming connections in popup windows, some failing)

```bash
% make test
Running tests...
Test project /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-transport/build
        Start   1: UNIT_AdvertiseOptions_TEST
  1/102 Test   #1: UNIT_AdvertiseOptions_TEST .............................   Passed    0.37 sec
        Start   2: check_UNIT_AdvertiseOptions_TEST
  2/102 Test   #2: check_UNIT_AdvertiseOptions_TEST .......................   Passed    0.02 sec
        Start   3: UNIT_CIface_TEST
  3/102 Test   #3: UNIT_CIface_TEST .......................................   Passed    0.16 sec
        Start   4: check_UNIT_CIface_TEST
  4/102 Test   #4: check_UNIT_CIface_TEST .................................   Passed    0.02 sec
        Start   5: UNIT_Clock_TEST
[... 417 lines cut ...]
 -- /foo2
 -- /bar1
 -- /bar2
[       OK ] recorder.BeginRecordingAllBeforeAdvertisement (2311 ms)
[ RUN      ] recorder.DataWriterQueue
[       OK ] recorder.DataWriterQueue (313 ms)
[ RUN      ] recorder.DataWriterQueueClockUpdates
[       OK ] recorder.DataWriterQueueClockUpdates (316 ms)
[ RUN      ] recorder.DataWriterQueueDefaultBufferSize
[       OK ] recorder.DataWriterQueueDefaultBufferSize (719 ms)
[ RUN      ] recorder.DataWriterQueueInfiniteBufferSize
[       OK ] recorder.DataWriterQueueInfiniteBufferSize (723 ms)
[ RUN      ] recorder.DataWriterQueueLargeMessages
[       OK ] recorder.DataWriterQueueLargeMessages (337 ms)
[----------] 8 tests from recorder (11137 ms total)

[----------] Global test environment tear-down
[==========] 8 tests from 1 test suite ran. (11137 ms total)
[  PASSED  ] 7 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] recorder.BeginRecordingTopicsAfterAdvertisement

 1 FAILED TEST
  YOU HAVE 1 DISABLED TEST
```

srmainwaring, 2023-09-13:
Re-run using `ctest -V`:

```bash
% ctest -V
UpdateCTestConfiguration  from :/Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-transport/build/DartConfiguration.tcl
Parse Config file:/Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-transport/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-transport/build/DartConfiguration.tcl
Parse Config file:/Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-transport/build/DartConfiguration.tcl
Test project /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-transport/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
        Start   1: UNIT_AdvertiseOptions_TEST

1: Test command: /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-transport/build/bin/UNIT_AdvertiseOptions_TEST "--gtest_output=xml:/Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-transport/build/test_results/UNIT_AdvertiseOptions_TEST.xml"
1: Working Directory: /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-transport/build/src
1: Test timeout computed to be: 240
1: Running main() from /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-transport/test/gtest_vendor/src/gtest_main.cc
1: [==========] Running 18 tests from 1 test suite.
1: [----------] Global test environment set-up.
1: [----------] 18 tests from AdvertiseOptionsTest
1: [ RUN      ] AdvertiseOptionsTest.defConstructor
1: [       OK ] AdvertiseOptionsTest.defConstructor (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.copyConstructor
1: [       OK ] AdvertiseOptionsTest.copyConstructor (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.assignmentOp
1: [       OK ] AdvertiseOptionsTest.assignmentOp (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.equalityOp
1: [       OK ] AdvertiseOptionsTest.equalityOp (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.streamInsertion
1: [       OK ] AdvertiseOptionsTest.streamInsertion (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.accessors
1: [       OK ] AdvertiseOptionsTest.accessors (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.msgDefConstructor
1: [       OK ] AdvertiseOptionsTest.msgDefConstructor (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.msgCopyConstructor
1: [       OK ] AdvertiseOptionsTest.msgCopyConstructor (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.msgAssignmentOp
1: [       OK ] AdvertiseOptionsTest.msgAssignmentOp (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.msgEqualityOp
1: [       OK ] AdvertiseOptionsTest.msgEqualityOp (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.msgStreamInsertion
1: [       OK ] AdvertiseOptionsTest.msgStreamInsertion (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.msgAccessors
1: [       OK ] AdvertiseOptionsTest.msgAccessors (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.srvDefConstructor
1: [       OK ] AdvertiseOptionsTest.srvDefConstructor (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.srvCopyConstructor
1: [       OK ] AdvertiseOptionsTest.srvCopyConstructor (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.srvAssignmentOp
1: [       OK ] AdvertiseOptionsTest.srvAssignmentOp (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.srvEqualityOp
1: [       OK ] AdvertiseOptionsTest.srvEqualityOp (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.srvStreamInsertion
1: [       OK ] AdvertiseOptionsTest.srvStreamInsertion (0 ms)
1: [ RUN      ] AdvertiseOptionsTest.srvAccessors
1: [       OK ] AdvertiseOptionsTest.srvAccessors (0 ms)
1: [----------] 18 tests from AdvertiseOptionsTest (0 ms total)
1: 
1: [----------] Global test environment tear-down
1: [==========] 18 tests from 1 test suite ran. (0 ms total)
1: [  PASSED  ] 18 tests.
  1/102 Test   #1: UNIT_AdvertiseOptions_TEST .............................   Passed    0.04 sec
test 2
        Start   2: check_UNIT_AdvertiseOptions_TEST

[... 2816 lines cut ...]
102: /opt/homebrew/Cellar/python@3.11/3.11.4_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/importlib/__init__.py:126: in import_module
102:     return _bootstrap._gcd_import(name[level:], package, level)
102: ../../python/test/options_TEST.py:1: in <module>
102:     from gz.msgs10.stringmsg_pb2 import StringMsg
102: E   ModuleNotFoundError: No module named 'gz'
102: - generated xml file: /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-transport/build/test_results/options_TEST.xml -
102: =========================== short test summary info ============================
102: ERROR ../../python/test/options_TEST.py
102: !!!!!!!!!!!!!!!!!!!! Interrupted: 1 error during collection !!!!!!!!!!!!!!!!!!!!
102: =============================== 1 error in 0.05s ===============================
102/102 Test #102: options_TEST.py ........................................***Failed    0.22 sec

96% tests passed, 4 tests failed out of 102

Total Test time (real) = 267.67 sec

The following tests FAILED:
	 47 - INTEGRATION_twoProcsSrvCall (Failed)
	100 - pubSub_TEST.py (Failed)
	101 - requester_TEST.py (Failed)
	102 - options_TEST.py (Failed)
Errors while running CTest
Output from these tests are in: /Users/rhys/Code/osrf/gz_harmonic_ws/src/gz-transport/build/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
```

azeey, 2023-09-14:
I think the python tests actually require running `make install` first.

azeey, 2023-09-20:
https://github.com/gazebosim/gz-transport/pull/441

azeey, 2023-09-25:
I think the `twoProcsSrvCall` test is flaky. There's already an issue for windows: https://github.com/gazebosim/gz-transport/issues/215

closed 2023-09-25 by azeey