gazebosim/gazebo_test_cases #945: gz-utils: Installation

Voldivh, 2023-09-11:
## Setup
- Os: MacOS
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/utils/2/install.html)
- [Source](https://github.com/gazebosim/gz-utils/tree/gz-utils2/tutorials/installation.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2023-09-13:
- I intend to build all libs from source and so will also be following https://gazebosim.org/docs/harmonic/install_osx_src

- The `colcon` build instructions suggest using dart 6.10 and setting the `CMAKE_PREFIX_PATH`, however the brew binary install uses the rolling version. The instructions should be updated.

- The `CMAKE_PREFIX_PATH` needs to be set for `qt@5`

- `qt` should be unlinked when building from source to avoid conflicts
```bash
brew unlink qt
```

- Cross check version and dependencies using

```bash
$brew info osrf/simulation/gz-harmonic
==> osrf/simulation/gz-harmonic: stable 1.0.0~pre2, HEAD
Collection of gazebo simulation software
https://github.com/gazebosim/gz-harmonic
Not installed
From: https://github.com/osrf/homebrew-simulation/blob/HEAD/Formula/gz-harmonic.rb
License: Apache-2.0
==> Dependencies
Build: cmake ✘, python@3.11 ✘
Required: gz-cmake3 ✘, gz-common5 ✘, gz-fuel-tools9 ✘, gz-gui8 ✘, gz-launch7 ✘, gz-math7 ✘, gz-msgs10 ✘, gz-physics7 ✘, gz-plugin2 ✘, gz-rendering8 ✘, gz-sensors8 ✘, gz-sim8 ✘, gz-tools2 ✘, gz-transport13 ✘, gz-utils2 ✘, pkg-config ✔, sdformat14 ✘
==> Options
--HEAD
	Install HEAD version
```

- Build dependencies

```bash
colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-select gz-cmake3
Starting >>> gz-cmake3
--- stderr: gz-cmake3                              
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_INSTALL_NAME_DIR
    CMAKE_MACOSX_RPATH

warning: Tag 'OUTPUT_TEXT_DIRECTION' at line 102 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'CLANG_ASSISTED_PARSING' at line 1072 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'CLANG_OPTIONS' at line 1080 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'CLANG_DATABASE_PATH' at line 1090 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'HTML_TIMESTAMP' at line 1237 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'FORMULA_TRANSPARENT' at line 1529 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'LATEX_SOURCE_CODE' at line 1837 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'LATEX_TIMESTAMP' at line 1853 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'RTF_SOURCE_CODE' at line 1927 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOCBOOK_PROGRAMLISTING' at line 2032 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'CLASS_DIAGRAMS' at line 2219 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOT_FONTNAME' at line 2261 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOT_FONTSIZE' at line 2268 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOT_TRANSPARENT' at line 2494 of file '/Users/rhys/Code/osrf/gz_harmonic_ws/build/gz-cmake3/api_tagfile.dox' has become obsolete.
[... cut ...]
% colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-select gz-tools2
Starting >>> gz-tools2
Finished <<< gz-tools2 [1.71s]                     

Summary: 1 package finished [1.90s]
```

- `gz-utils2` builds clean

```bash
% colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --packages-select gz-utils2
Starting >>> gz-utils2
Finished <<< gz-utils2 [1.68s]                     

Summary: 1 package finished [1.87s]
```

mjcarroll, 2023-09-13:
> The doxygen api.in file could be cleaned up for building on macOS. There are a number of options that are either deprecated or not compiled into the brew version of doxygen.

This is true for non-macOS platforms as well. `utils` has to support bionic-jammy, which is a large spread of doxygen versions.  It's been discussed to address it, but hasn't been a high priority.

srmainwaring, 2023-09-13:
@mjcarroll - no prob. I maintain patches for my own dev environment. Had thought it may have been macOS specific.

azeey, 2023-09-25:
I'll go ahead and close this since the only issue is doxygen warnings, which I'm not sure if we can easily fix.

closed 2023-09-25 by azeey

azeey, 2023-09-26:
Hi @srmainwaring, congratulations! You are one of our top contributors in the Harmonic tutorial party. Can you send your email address to `addisu@openrobotics.org`  so we can send you the Harmonic t-shirt?

azeey, 2023-11-16:
> Hi @srmainwaring, congratulations! You are one of our top contributors in the Harmonic tutorial party. Can you send your email address to `addisu@openrobotics.org`  so we can send you the Harmonic t-shirt?

Hi @srmainwaring, I don't think I ever received your email. If you're still interested in the Harmonic t-shirt, please email me.