gazebosim/gazebo_test_cases #1044: docs: moving_robot

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/ionic/moving_robot)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/moving_robot.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-09-02:
Interested

Mechazo11, 2024-09-04:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco full desktop
    ```
* **Summary**
    * Works. When ```Key Press``` plugin is added from the ```Gazebo Sim``` drop down menu, ```(2024-09-03 21:44:22.780) [error] [GUI] Add [KeyPublisher] via menu``` error is seen.

https://github.com/user-attachments/assets/92a169da-bbe4-4b97-b18d-791beecd14c4

* **Suggestion**
    * Before calling ```gz topic -e -t /keyboard/keypress```, the tutorial should state how to activate the Triggered publisher. 
    * The fact that ```TriggeredPublisher``` should be added into ```building_robot.sdf``` should be mentioned in the tutorial.

Mechazo11, 2024-09-04:
@azeey-test this may be closed.

iche033, 2024-09-04:
I ticketed an issue for the error msg. https://github.com/gazebosim/gz-common/issues/632

closed 2024-09-04 by iche033