gazebosim/gazebo_test_cases #1045: docs: moving_robot

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/docs/ionic/moving_robot)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/moving_robot.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

knmcguire, 2024-09-04:
You can assign me to this issue

knmcguire, 2024-09-05:
Success! ✅ with a couple of warning/errors

Opening up the keypublisher plugin on the gui did give an error though, but it seemed to work none the less:
```
gz sim -g
DynamicFactory(): Unable to parse descriptor set from [C:/Users/kimbe/Development/simulation/gazebo/gz-ws/install\share\gz\protos\gz-msgs11.gz_desc]
(2024-09-05 09:41:56.802) [error] [MainWindow.cc:268] [GUI] Add [KeyPublisher] via menu
```

also terminating the gui gave a stack trace (forgot to mention that with the previous issues.
```
Stack trace (most recent call last):
#9    Object "", at 00007FFC2526257D, in BaseThreadInitThunk
#8    Object "", at 00007FFB8F125FA1, in QThread::start
#7    Object "", at 00007FFBAC1E2695, in QSGOpenGLLayer::updateTexture
#6    Object "", at 00007FFBAC1E1D9D, in QSGOpenGLLayer::updateTexture
#5    Object "", at 00007FFBAC1DF0F6, in QSGOpenGLLayer::updateTexture
#4    Object "", at 00007FFBAC1E0C6C, in QSGOpenGLLayer::updateTexture
#3    Object "", at 00007FFBAC221C40, in QQuickWindowPrivate::cleanupNodesOnShutdown
#2    Object "", at 00007FFBAC22199B, in QQuickWindowPrivate::cleanupNodes
#1    Object "", at 00007FFBAC183CD3, in operator<<
#0    Object "", at 00007FFBAC183F00, in QSGNode::destroy
```

But it all didn't block me to get to end result:

https://github.com/user-attachments/assets/b06ec97b-adbe-4077-b4bc-082b5566634d

iche033, 2024-09-05:
ticketed an issue for the error msg: https://github.com/gazebosim/gz-common/issues/632

knmcguire, 2024-09-10:
Thanks! I no longer get the error message (the key publisher one). 

The stack trace I still get though when closing the gui side

azeey, 2024-09-13:
@knmcguire mind opening an issue for the crash in exit problem? I'll go ahead and close this since the instructions work.

closed 2024-09-13 by azeey

knmcguire, 2024-09-14:
Done! I've done it in gz-sim so I hope that is the right place: https://github.com/gazebosim/gz-sim/issues/2623