gazebosim/gazebo_test_cases #1065: docs: hotkeys

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/ionic/hotkeys)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/hotkeys.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-09-02:
Interested in !

avanmalleghem, 2024-09-02:
Not working
- In MODE : Esc (Selection mode), r (rotate), t (translate)
- In MANIPULATING MODELS : r, t
- Does x/y/z still makes sense ? We can only move along one axis at a time, right ?

Expected or not ? If yes, I can create a PR.

iche033, 2024-09-04:
oh I think we lost those functionality when we reimplemented the 3D scene gui plugin. I think we should remove those entries since they are not working yet and we may not implement them soon. Can you create a PR? Thanks.

avanmalleghem, 2024-09-05:
PR opened : https://github.com/gazebosim/docs/pull/500

ahcorde, 2024-09-06:
PR is merged, thank you for testing!

closed 2024-09-06 by ahcorde