gazebosim/gazebo_test_cases #1080: gz-cmake: README

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/libs/cmake)
* [Source](https://github.com/gazebosim/gz-cmake/blob/gz-cmake4/README.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-09-02:
Interested in !

avanmalleghem, 2024-09-03:
- code of conduct and contribution (order in table of content + link broken)
- contributing.md -> contribution guide
- Typo "C+±based"
- "projects’"
- [Examples](https://gazebosim.org/libs/cmake/#examples/) is incorrect. Same for "Tutorials"
- "[MIGRATION.md guide](https://github.com/gazebosim/gz-cmake/blob/master/MIGRATION.md)." broken link

Will create a PR soon

avanmalleghem, 2024-09-03:
PR done : https://github.com/gazebosim/gz-cmake/pull/449

ahcorde, 2024-09-04:
PR is merged, thank you for testing!

closed 2024-09-04 by ahcorde