gazebosim/gazebo_test_cases #1083: gz-cmake: Developing with Gazebo CMake

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/cmake/4/developingwithcmake.html)
* [Source](https://github.com/gazebosim/gz-cmake/tree/gz-cmake4/tutorials/developing_with_gz-cmake.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-09-02:
Interested

avanmalleghem, 2024-09-05:
Interested in !

avanmalleghem, 2024-09-06:
- Link [Check the documentation for `GZ_SANITIZER` flag](https://gazebosim.org/api/cmake/4/gz__cmake__sanitizers_8md.html) is dead
- Typo "to disables tests"

Will open a PR soon.

avanmalleghem, 2024-09-06:
PR available here : https://github.com/gazebosim/gz-cmake/pull/452

ahcorde, 2024-09-06:
Thank you for testing!

closed 2024-09-06 by ahcorde