gazebosim/gazebo_test_cases #1089: gz-cmake: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/cmake/4/install.html)
* [Source](https://github.com/gazebosim/gz-cmake/tree/gz-cmake4/tutorials/install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

trushant05, 2024-08-29:
Everything works as expected for both **Source Installation** and **Binary Installation**.

trushant05, 2024-08-29:
## Binary Installation
Everything works fine

## Source Installation
Everything works fine

## Documentation
Everything works fine

## Testing
Everything works fine

closed 2024-08-30 by iche033